openwrt/luci #2613: A wireless device cannot have it's mac address set from luci (option macaddr)

heydojo, 2019-03-03:
After installing the OpenWrt firmware into a Sagem F@ST2704 V2, the MAC address for the wireless device needs to be manually set by establishing a connection into the device via ssh and editing the file /etc/config/wireless before restarting the wifi. Otherwise wifi does not work.

https://openwrt.org/start?id=toh/sagem/fast2704

It should IMO be possible to manually set the MAC address for a wifi adapter using luci. So there is no need to ssh in but it is not.

Please could a new field be added to the Wireless Network: Device Configuration - Advanced settings tab, which allows `option macaddr` to be optionally manually set?

For reference, /etc/config/wireless needs to look something like this:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11bg'
        option path	'pci0000:00/0000:00:00.0/0000:01:00.0/bcma0:1'
	option country 'GB'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option macaddr 'xx:xx:xx:xx:xx:xx'
	option wmm '0'
```

I think that https://github.com/openwrt/luci/issues/1100#issuecomment-464374463 might be related to this issue but the report is from 2017.

Thanks

heydojo, 2019-03-07:
Override MAC address can confusingly be found in Network - Interfaces - WWAN (if wwan is the interface for the wifi) - Advanced Settings.

For example: http://openwrt/cgi-bin/luci/admin/network/network

So my question now is : Is the config on the wiki wrong where device settings are being set in the interface configuration section?

jow-, 2019-03-07:
OpenWrt / netifd supports multiple ways to set a mac address. Historically there's been option macaddr in the config interface section, this is what LuCI exposes. Downside is that this MAC is applied to the bridge interface in case multiple interfaces are bridged together.

To address this. OpenWrt eventually introduced "config device" sections to set things like macaddr, queue lengths etc. per netdev. LuCI currently does not support adding or modifying "config device" sections at all. Support for this needs to be implemented first.

heydojo, 2019-03-09:
Thanks for the reply @jow-.

It just seemed odd to me.

closed 2019-03-09 by heydojo