openwrt/luci #2624: Password input rendered wrong when embedded in table cell

yousong, 2019-03-15:
The issue was reported as a followup in Below is a copy of comment https://github.com/openwrt/luci/issues/2398#issuecomment-450495221 , but it got lost there after the original issue got resolved and closed

Below is a copy of that comment.  The issue is still there in current luci master.

> I understand.  Previously browser autocomplete can indeed cause trouble.  But it seems when embedded in a table, the added input element will be visible, despite the css code `left: -1000px`
> 
> [image]
> 
> Another minor undesirable change is that previously the `Reveal/hide password` button stick to the input element on the same row, not wrapping to the next line.
> 
> [image]

shNanChen, 2019-06-25:
This problem still exists. :flushed:

jow-, 2019-06-25:
Wasn't able to reprioduce this in Firefox.

yousong, 2019-06-25:
I can confirm that it's still reproducible with a recent version of Chrome and Safari.  Sorry that I am still not certain when I can allocate time for this.  I will update here when I am on it.

yousong, 2019-06-28:
I just opened a pull request for this.  After the change, the result will be like the following

[image]

PR #2803 (2019-07-06): luci-base: fix display of password input
Should fix #2624
closed 2019-07-06 by yousong