openwrt/luci #2625: [Feature Request] per IP / port / device / interface active connections monitoring

TheHellSite, 2019-03-15:
I really like the active connections monitoring page in LuCi. 
However I think it would nice if we could apply some filters to this page.

When there are a lot of connections happening on the router the firefox tab viewing the connections always freezes.
By filtering for the ip / port / device / interface we are interested in, this could be prevented and would also make it easier to view the data.

baker-Xie, 2019-05-23:
@TheHellSite  I'm agree with it. In addition , I think it would be nice if we could apply some additional function like following:
1. Display all the connect device (including static IP address)
2. Add a icon to show the device categories like cellphone / computer / TV and so on.
3. Add a button that user can modify the device name/category
4. Add a button to block the device that it cannot access the internet
5. Display the received/send data amount of each IP
I'm just write my advice from a users' point of view.
Thanks for all you have done.

rcmonitor, 2019-06-01:
++

skupjoe, 2021-03-13:
Please support this! I'd like a quick and easy way of seeing all wireless/dhcp/static IPs that are connected in one place. 

To do this currently, I look at ARP routes, but this can contain outdated information/clients that are no longer active.