openwrt/luci #2626: wifi tx power level doesn't show what is in the config file.

kroeckx, 2019-03-15:
The allowed power level for the wifi depends on selected channel. At least where I live, some channels allow 20, others allow 27. Depending on the channel that it currently selected, it seems the drop down box updates it's list of allowed power levels. I have the channel set to auto. At some point it was at a channel that allows 27, and I set it to 27 at that point. At some point later, it was at a different channel, and it showed that the setting was "auto", while the config file actually has 27 in it. I wanted to change it to auto, but since it was already at auto, it didn't actually change my config.

This is with OpenWRT 18.06.2

silence48, 2019-08-11:
when i update any power level it doesn't add it to config, just always says auto i gotta do it manually

feckert, 2019-09-19:
There were a lot of changes in the Luci.
Lua code was replaced by Javascript code.
Could you please try the last snapshot probably it works again now.
If not I will have a look.

closed 2025-01-10 by systemcrash