openwrt/luci #2628: wireless: invalid wireless modes are listed on the config page

pmelange, 2019-03-18:
With many of the newer routers, adhoc mode is not supported in the 5Ghz devices.  Configuring the device with an invalid mode naturally doesn't work.

For example, on a ubnt-unifiac-mesh:

[image]

When running ```iw phy ph0 info``` on the ubnt-unifiac-mesh the following results are shown:
```
	valid interface combinations:
		 * #{ AP, mesh point } <= 8, #{ managed } <= 1,
		   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }
```

It would be good to have LuCI provide options which will actually work.

feckert, 2019-09-19:
What version of openwrt do you use?
Or is this a problem of the freifunk-berlin/firmware-package?
If so please close the issue it was already related and merged.

pmelange, 2019-09-19:
Please re-open this issue (i don't have permission). 

 It is still a problem with the latest snapshot.  The device I am using is an Arch C7 V4.  Snapshot image r11037-1979af10ca with LuCI installed.

The AC device does *not* support 802.11s mesh, yet in the web interface, 802.11s is listed as an option.

iw list 
```
	valid interface combinations:
		 * #{ managed, P2P-client } <= 16, #{ P2P-GO } <= 3, #{ AP } <= 16, #{ IBSS } <= 1,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }
```

Screenshot from LuCI
[image]

ironvegeta, 2020-11-26:
I can still see this issue. 802.11s is also selectable and leads to an error on "wireless security" > Encryption
[image]

Is there any date when 802.11s is supported in the ui?

 Powered by LuCI 
* openwrt-19.07 branch (git-20.287.57033-3d52019) / OpenWrt 19.07.4 r11208-ce6496d796
* package luci	git-20.287.57033-3d52019-1
* package wpad-wolfssl - 2019-08-08-ca8c2bd2-4

Ansuel, 2022-06-07:
@feckert @jow- why this is not fixed? i mean shouldn't be that hard to get the supported modes and report a correct list

jow-, 2022-06-07:
Lack of time and low severity