openwrt/luci #2631: luci-app-wifischedule - Save & Apply not working properly

CaboBeach, 2019-03-19:
Save & Apply usually has to be clicked two or more times in Services > Wifi schedule > Schedule for settings to take affect.

dibdot, 2019-03-19:
Please test attached PR - thanks!

CaboBeach, 2019-03-19:
Thanks dibdot, sorry to ask, but could you please let me know the path to `/applications`? Documentation says `luci-path/applications`, but I'm not sure what `luci-path/` should be unfortunately.

dibdot, 2019-03-19:
You should find the file on your router under `/usr/lib/lua/luci/model/cbi/wifischedule/...`

CaboBeach, 2019-03-19:
Thanks again dibdot, unfortunately the settings don't seem to be getting applied after the change, they no longer show in cron and wifi stays up, here's the relevant section of my `/wifi_schedule.lua` just to make sure I made the correct change - 

```
-- BEGIN Map
m = Map("wifi_schedule", translate("Wifi Schedule"), translate("Defines a schedule when to turn on and off wifi."))
function m.on_apply(self)
    sys.exec("/usr/bin/wifi_schedule.sh cron")
end
-- END Map
```

dibdot, 2019-03-19:
Sorry, I forgot to set the 'apply_on_parse' attribute on this map instance. I've updated the PR - should work now.

CaboBeach, 2019-03-19:
That got it, it's working perfect now, many thanks as always for your hard work dibdot.

closed 2019-03-20 by CaboBeach

dibdot, 2019-03-20:
Thanks for testing, will merge!