openwrt/luci #2633: OPKG install fails when called from Luci

arturioxas, 2019-03-20:
Hello, 

I succesfully used various opkg calls to install, update, remove packages, etc. Have never seen any faults, yet calling the same functions from luci interface generates a segmentation fault. No package can be installed but I can remove existing ones. Maybe kernel panic log might help:
```
Tue Mar 19 15:31:51 2019 daemon.err lighttpd[18201]: (server.c.1828) server stopped by UID = 0 PID = 1 
Tue Mar 19 15:31:52 2019 daemon.err lighttpd[23625]: (log.c.217) server started 
Tue Mar 19 15:32:12 2019 kern.info kernel: [ 7063.156366] 
Tue Mar 19 15:32:12 2019 kern.info kernel: [ 7063.156366] do_page_fault(): sending SIGSEGV to opkg for invalid read access from 00000000
Tue Mar 19 15:32:12 2019 kern.info kernel: [ 7063.156396] epc = 7791c9a0 in libc.so[778aa000+91000]
Tue Mar 19 15:32:12 2019 kern.info kernel: [ 7063.156450] ra  = 778c8a28 in libc.so[778aa000+91000]
Tue Mar 19 15:32:12 2019 kern.info kernel: [ 7063.156475] 
Tue Mar 19 15:32:30 2019 kern.info kernel: [ 7081.010839] 
Tue Mar 19 15:32:30 2019 kern.info kernel: [ 7081.010839] do_page_fault(): sending SIGSEGV to opkg for invalid read access from 00000000
Tue Mar 19 15:32:30 2019 kern.info kernel: [ 7081.010965] epc = 776ca9a0 in libc.so[77658000+91000]
Tue Mar 19 15:32:30 2019 kern.info kernel: [ 7081.011018] ra  = 77676a28 in libc.so[77658000+91000]
Tue Mar 19 15:32:30 2019 kern.info kernel: [ 7081.011044] 
``` 
As one can see, failure happens at different addresses. I suspected stack problems but couldn't succesfully trace it. My libc is of version `1.1.16-1` (musl-libc), opkg version `9c97d5ecd795709c8584e972bfdf3aee3a5b846d-16`, using ar71xx architecture.

jow-, 2019-03-20:
How do you call opkg install from LuCI? Do you run LuCI under uhttpd or lighttpd?

arturioxas, 2019-03-20:
I'm running from lighttpd. The function it breaks on is:
```
-- Internal action function
local function _action(cmd, ...)
        local pkg = ""
        for k, v in pairs({...}) do
                pkg = pkg .. " '" .. v:gsub("'", "") .. "'"
        end
        local c = "%s %s %s >/tmp/opkg.stdout 2>/tmp/opkg.stderr" %{ ipkg, cmd, pkg }
        local r = os.execute(c) -- breaks here
        local e = fs.readfile("/tmp/opkg.stderr")
        local o = fs.readfile("/tmp/opkg.stdout")
        
        fs.unlink("/tmp/opkg.stderr")
        fs.unlink("/tmp/opkg.stdout")
        return r, o or "", e or ""
end
```

jow-, 2019-03-20:
Which OpenWrt release? Can you reproduce this on uhttpd as well?

arturioxas, 2019-03-20:
Seems like the problem went away with a newer release that I compiled and installed manually. Closing the thread.

closed 2019-03-20 by arturioxas