openwrt/luci #2635: LuCi Rosy Enhancement Request

davidc502, 2019-03-21:
Enhancement Request for LuCi Rosy

Replace word LEDE with OpenWrt for the Splash Screen and General pages.

dibdot, 2019-03-22:
@rosysong

rosysong, 2019-03-24:
@davidc502  Could you please show us where is the word "LEDE" in LuCI Rosy theme ?

hnyman, 2019-03-24:
> Replace word LEDE with OpenWrt for the Splash Screen and General pages.

Likely the same old thing again: It is not branding, but hostname.
The router's config has LEDE as the host name in /etc/config/system (from the old default config). That gets shown on the upper left corner.

LuCI source code has no "LEDE" word in the rosy theme.

I have named my own router as "router1", so it says that instead of "OpenWrt":

[image]

dibdot, 2019-03-29:
no more feedback from @davidc502  - seems to be fixed.

closed 2019-03-29 by dibdot

davidc502, 2019-03-29:
Sorry.. I haven't kept good track of this non-issue.. confirmed -- Not a branding issue, but where LuCi pulls the name from which is the system file.  I changed the option hostname lede to OpenWrt, and after a reboot LuCi displays OpenWrt.  

thanks for the help.