openwrt/luci #2645: luci-theme-rosy: Network/Switch view does not fit the screen

fantom-x, 2019-03-30:
Built from master yesterday and using the latest Firefox on MacOs. A half of the view does not fit and there is no horizontal scroll bar.

[image]

ping @rosysong

rosysong, 2019-04-07:
@fantom-x  
Fixed, please try below PR, thanks!
https://github.com/openwrt/luci/pull/2663

closed 2019-04-07 by hnyman

fantom-x, 2019-04-13:
@rosysong I just compiled OpenWRT master @ 9385ff654e and the issue is still present.

rosysong, 2019-04-13:
Will updated it later, this issue seems occur on routers with 5 ports.

fantom-x, 2019-04-26:
Any luck?

rosysong, 2019-04-27:
@fantom-x  Please wait patiently, thanks!

fantom-x, 2019-05-05:
Thx for the fix, the table can now be scrolled horizontally. But it is still somewhat inconvenient to use: is it possible to shrink the table so that it is possible to see the entire table at once?

commit df629699 (2019-04-07): luci-theme-rosy: fix #2645

fix view on network / switch

https://github.com/openwrt/luci/issues/2645