openwrt/luci #2646: Drop-down height needs to be larger

David-Hari, 2019-03-31:
The drop-down boxes have ```max-height: 130px```. I need to scroll to see more items, as in this example:
[image]

I have searched the code and cannot find where the height is being set, but it's done so on the element itself so I cannot change via CSS (well I could, but I don't want to use !important).

Is there any reason why the height is so small? Could it be changed to something like 250px or 300px?

jow-, 2019-03-31:
Yes it is intentional. It can be increased by adjusting the `display-items` property.

closed 2019-03-31 by jow-

David-Hari, 2019-04-01:
Thanks for the reply.

Two questions:
1. You say it is intentional. May I ask why?
2. Where is this "display-items" property? It does not appear to be a CSS thing. Is it in some LuCI config file?

jow-, 2019-04-01:
1) It is intended to mimic the behaviour of a normal select box, these do not grow indefinitely in height either but expand towards a max size, then present a scroll bar to reach further items.

2) https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/view/cbi/network_ifacelist.htm#L44

David-Hari, 2019-04-02:
I think you misunderstand what I am asking for. It's not that I want the height to grow indefinitely, it's just that I think a max height of 130px is woefully small. Why can't it be 250px or 300px as I asked for?

jow-, 2019-04-02:
It can, if you change the display-items value of the template from 5 to e.g. 10. The max height is the product of the display-items value and the individual height of the first option element.

David-Hari, 2019-04-02:
Ok, well I was asking you to do that. I would have submitted a pull request, but I did not know what needed to be changed.

David-Hari, 2019-04-02:
Thanks. Perhaps also increase the default size in ```cbi.js```.

commit fb453f55 (2019-04-02): luci-base: increase max size of network dropdowns

Fixes: #2646