openwrt/luci #2658: wndr4300刷了OpenWrt R8.1.7 By Lean 交管12123提示网络请求失败

donny861221, 2019-04-04:
wndr4300刷了OpenWrt R8.1.7 By Lean  交管12123APP提示网络请求失败怎么回事？
我用手机卡电信 联通的4G都没问题，可以登录
还有我在外网连上wndr4300的PPTP vpn后连接速度最高也就1MB/s，以前都是在3MB/s的
请问是怎么回事?

hnyman, 2019-04-04:
English please.
And likely not related to the LuCI user interface.

closed 2019-04-04 by hnyman

donny861221, 2019-04-04:
Wndr4300 swiped OpenWrt R8.1.7 By Lean traffic 12123APP prompt network request failed what happened?

I use mobile phone card the 4G of telecommunication unicom is no problem, can login

In addition, after I connected to the PPTP VPN of wndr4300 on the external network, the connection speed was only 1MB/S, which was always 3MB/S before

hnyman, 2019-04-06:
Again: nothing you describe is about the LuCI user interafce, which this openwrt/luci site is about.

And there is no official "OpenWrt R8.1.7 By Lean". That sounsd like somebody's private compilation, so you better ask them for support. (That 8.1.7 gives no indication about the actual underlying OpenWrt version, so pretty impossible to comment anything)