openwrt/luci #2659: statistics: Negative average traffic

davean, 2019-04-04:
As in the closed issue #2302 I see negative average traffic frequently. All I need to do to reproduce it is open the realtime traffic graph, transfer a file, and wait a few minutes. It reliably happens within half an hour. It comes along with a break in the traffic graph.

For example: 
[image]

jow-, 2019-04-05:
Please capture and paste the output of `luci-bwc -i eth1` via SSH once you see the negative average in the screen, that should simplify reproducing this.

davean, 2019-04-05:
This was at a higher transfer rate, it went negative, recovered and then went negative again.

Right after the first negative:

[ 1554480386, 2638177925, 2175612905, 462911266, 1136888931 ],
[ 1554480388, 2638183599, 2175612920, 462912799, 1136888947 ],
[ 1554480389, 2638188543, 2175612934, 462914735, 1136888967 ],
[ 1554480391, 2638190776, 2175612949, 462916196, 1136888981 ],
[ 1554480392, 2638193588, 2175612965, 462918425, 1136889002 ],
[ 1554480394, 2638200583, 2175612978, 462919169, 1136889012 ],
[ 1554480395, 2638207759, 2175613014, 462922578, 1136889047 ],
[ 1554480397, 2638210259, 2175613026, 462923564, 1136889058 ],
[ 1554480398, 2638214809, 2175613054, 462926567, 1136889086 ],
[ 1554480400, 2638218949, 2175613081, 462928819, 1136889107 ],
[ 1554480401, 2638222543, 2175613107, 462931971, 1136889134 ],
[ 1554480403, 2638225813, 2175613129, 462933721, 1136889148 ],
[ 1554480404, 2638229502, 2175613154, 462936528, 1136889171 ],
[ 1554480406, 2638231541, 2175613166, 462937376, 1136889178 ],
[ 1554480407, 2638232771, 2175613169, 462937586, 1136889181 ],
[ 1554480409, 2638233300, 2175613172, 462937776, 1136889183 ],
[ 1554480410, 2638235417, 2175613183, 462938754, 1136889195 ],
[ 1554480412, 2638236519, 2175613189, 462939305, 1136889202 ],
[ 1554480413, 2638240260, 2175613204, 462940329, 1136889216 ],
[ 1554480415, 2638242012, 2175613214, 462942275, 1136889225 ],
[ 1554480416, 2638246928, 2175613246, 462946521, 1136889255 ],
[ 1554480418, 2638250205, 2175613268, 462948610, 1136889272 ],
[ 1554480419, 2638257111, 2175613312, 462952059, 1136889305 ],
[ 1554480421, 2638258564, 2175613322, 462953309, 1136889316 ],
[ 1554480422, 2638262163, 2175613331, 462954202, 1136889325 ],
[ 1554480424, 2638264073, 2175613336, 462954537, 1136889329 ],
[ 1554480425, 2638265397, 2175613340, 462954810, 1136889332 ],
[ 1554480427, 2638266002, 2175613343, 462955079, 1136889335 ],
[ 1554480428, 2638267076, 2175613348, 462955359, 1136889339 ],
[ 1554480430, 2638270575, 2175613361, 462959183, 1136889354 ],
[ 1554480431, 2638271820, 2175613367, 462959783, 1136889361 ],
[ 1554480433, 2638272149, 2175613369, 462959969, 1136889363 ],
[ 1554480434, 2638272593, 2175613371, 462960039, 1136889364 ],
[ 1554480436, 2638273445, 2175613376, 462960407, 1136889368 ],
[ 1554480437, 2638274797, 2175613382, 462960675, 1136889372 ],
[ 1554480439, 2638277976, 2175613395, 462961641, 1136889385 ],
[ 1554480440, 2638280250, 2175613408, 462962699, 1136889398 ],
[ 1554480442, 2638280880, 2175613412, 462963217, 1136889404 ],
[ 1554480443, 2638285013, 2175613436, 462967905, 1136889431 ],
[ 1554480445, 2641971587, 2175615921, 463080205, 1136889919 ],
[ 1554480446, 2694064051, 2175650532, 463557612, 1136895460 ],
[ 1554480448, 2732778658, 2175676511, 464076466, 1136899707 ],
[ 1554480449, 2764593880, 2175697966, 464507230, 1136903176 ],
[ 1554480451, 2765060541, 2175698299, 464513713, 1136903255 ],
[ 1554480452, 2854416944, 2175757629, 465119337, 1136911253 ],
[ 1554480454, 2938055284, 2175813755, 465667607, 1136918656 ],
[ 1554480455, 3090384697, 2175916258, 466670325, 1136932199 ],
[ 1554480457, 3179650795, 2175976287, 467263975, 1136940222 ],
[ 1554480458, 3318859560, 2176069894, 468177867, 1136952565 ],
[ 1554480460, 3406867750, 2176129061, 468756105, 1136960374 ],
[ 1554480461, 3553922283, 2176228031, 469722703, 1136973422 ],
[ 1554480463, 3647222757, 2176290824, 470340247, 1136981766 ],
[ 1554480464, 3784331732, 2176383054, 471242477, 1136993929 ],
[ 1554480466, 3875475733, 2176444334, 471847189, 1137002099 ],
[ 1554480467, 4024707349, 2176544692, 472827593, 1137015339 ],
[ 1554480469, 4114626826, 2176605103, 473413602, 1137023245 ],
[ 1554480470, 4258368365, 2176701792, 474352905, 1137035926 ],
[ 1554480472, 44815312, 2176756511, 474880794, 1137043044 ],
[ 1554480473, 194502496, 2176857253, 475863196, 1137056311 ],
[ 1554480474, 227007324, 2176879051, 476072052, 1137059127 ]

Right after the second:

[ 1554480466, 3875475733, 2176444334, 471847189, 1137002099 ],
[ 1554480467, 4024707349, 2176544692, 472827593, 1137015339 ],
[ 1554480469, 4114626826, 2176605103, 473413602, 1137023245 ],
[... 42 lines cut ...]
[ 1554480527, 3990243642, 2179409062, 501302057, 1137396172 ],
[ 1554480528, 4054550113, 2179452552, 501883426, 1137403143 ],
[ 1554480529, 4134027069, 2179506170, 502492506, 1137410941 ],
[ 1554480530, 4185123823, 2179540483, 502902388, 1137415476 ],
[ 1554480531, 4270647014, 2179597950, 503537529, 1137423096 ],
[ 1554480532, 65788914, 2179658558, 504120617, 1137430964 ],
[ 1554480533, 117237471, 2179693114, 504460262, 1137435539 ],
[ 1554480534, 207612101, 2179753845, 505058554, 1137443621 ],
[ 1554480535, 288517633, 2179808204, 505583626, 1137450707 ],
[ 1554480536, 344087837, 2179845620, 505953884, 1137455712 ],
[ 1554480537, 424517440, 2179899669, 506483492, 1137462864 ],
[ 1554480538, 517520882, 2179962242, 507093142, 1137471095 ],
[ 1554480539, 559600044, 2179990487, 507371462, 1137474855 ],
[ 1554480540, 646784425, 2180049126, 507939168, 1137482518 ],
[ 1554480541, 681530978, 2180072477, 508166193, 1137485578 ]

After the second one I managed to capture another screenshot:

[image]

It kept going negative and showing the drop out repeatedly, as you can see in a longer view its quite periodic when it drops out a second of the graph:

[image]

I'm actually a bit confused because the only time the graph might not be broken for me is when it is showing idleness (You can see that basically as soon as I open the graph it starts dropping out time periods and giving non-sensible results). I'm curious what is different for other people that this works at all for them.

davean, 2019-04-05:
And for the record, that is only a gigabit link. Other parts of the network are faster, but no ports on this device are. So its Peak value is bogus, as from all other observations is the spikiness. The negative values are hardly the only things that don't sanity check here.

davean, 2019-04-05:
An interesting side observation, opening up the eth0 and eth1 graphs next to each other, they mostly mirror each other except that the drop outs are not mirrored. There are occasionally holes in one but not the other graph and vice versa.

davean, 2019-04-05:
Well, the spikes were bothering me so I dug into it more.

I mirrored both the inbound and outbound ports and ran packet capture on them. Neither side had rate inconsistencies like are show in the graphs, nor were latencies through the device long. It seems that the variability of transfer rate shown in the graphs is also erroneous.

jow-, 2019-04-23:
I've been looking into it now, thanks to the data you've provided, the issue became apparent pretty fast. It boils down to the fact that the `luci-bwc` backend uses 32bit values for the traffic counters, which will wrap around pretty fast at high transfer rates.

Will add a fix for this soon.

commit e50edeeb (2019-07-19): luci-mod-status: luci-bwc: use 64bit traffic counters

Fixes: #2659
closed 2019-07-19 by jow-