openwrt/luci #2660: Changing LAN IP from LuCI makes the router unreachable

grinderx, 2019-04-06:
Hello,

I just flashed my Netgear WNDR3700v4 with the latest stable (18.06.2) OpenWrt. Changing the LAN IP from within LuCI from the default 192.168.1.1 to something like 192.168.20.2 (or whatever) for example results for the router to be unreachable. I 've tried twice and the only workaround is to ssh to the router and do the change from the CLI by directly editing /etc/config/network.

Related topics in the forum:

https://forum.openwrt.org/t/solved-how-to-correctly-change-router-lans-ip-address/21654
https://forum.openwrt.org/t/changing-lan-ip-unsuccesful/21819

Thanks,
G

jow-, 2019-04-06:
You need to change your PC LAN IP to 192.168.20.x after you changed the IP.

grinderx, 2019-04-06:
Correct! I forgot to mention that I have done the related change to my PC so that they are both in the same subnet when trying this.

jow-, 2019-04-06:
Did you wait for the rollback to occur, then click "Apply anyway" ? Otherwise LuCI will revert back to 192.168.1.1 while you try to each on the new IP, leading to the perception of an unreachable router.

grinderx, 2019-04-06:
Yes, I did wait. As I said, applying the same LAN IP changes from the CLI worked fine. It's exactly as described in the two forum threads I provided above.

jow-, 2019-04-06:
I know that it works fine from the cli, but that does not answer my question. How long did you wait exactly?

grinderx, 2019-04-06:
~30" Any other change I did from within LuCI worked.

jow-, 2019-04-06:
I skimmed the two forum threads but they do not seem helpful. I cannot reproduce the problem here either. Can you try again with the browser debugger (F12) open and observe the network tab?

grinderx, 2019-04-06:
I 'll try and do that, but it won't be any time soon (it's very hectic the last couple of weeks in our house). 

I use OpenSUSE Linux. and the WAN is on a different subnet, but I don't have the WAN port connected. I got that Netgear to use it purely as an AP and I noticed that behaviour while trying to set it up the way I want after flashing OpenWrt. I had my laptop directly connected to the router's LAN port 1 all the time while doing so.

adverserath, 2019-06-20:
On a clean install log into the router through http://openwrt, change the lan ip, quickly disable and enable your PC lan adapter, the change will complete.
Going in via the host name keeps the session active, the session is lost using the ip and you'll typically end up rolling back

RobGray-au, 2019-09-15:
or config the altenate IP on the network adaptor to be on the expected new network. so you can open the config page quickly.

toien, 2020-04-19:
i met the same problem and fixed it by follow **Command line instructions** on page: 

[OpenWrt as router device](https://openwrt.org/docs/guide-user/network/openwrt_as_routerdevice)

**Luci instructions** not work

rfmarves, 2020-06-25:
> On a clean install log into the router through http://openwrt, change the lan ip, quickly disable and enable your PC lan adapter, the change will complete.
> Going in via the host name keeps the session active, the session is lost using the ip and you'll typically end up rolling back

I had exactly this problem and your suggestion worked perfectly.
Steps:
1. Configured the IP change using LuCI
2. Clicked on Save and Apply
3. Unplugged the ethernet cable between the PC and the router after about 10 seconds. (I was using wired connection)

Worked perfectly!

tka85, 2021-05-03:
FWIW doing it from command line works as expected. 

Ssh into your device and assuming you want to move OpenWrt's LAN to the 192.168.2.x subnet:

# uci set network.lan.ipaddr='192.168.2.1' ; uci commit network ; service network restart

Restart device just for good measure and try ssh-ing again using the new IP.

Neustradamus, 2021-10-20:
I think that the network must be shut down after validation and start with the new network and the connected computers with DHCP must be automatically connected with the new network.

shoulders, 2021-11-23:
This is a better way: https://openwrt.org/faq/change_lan_ip

recolic, 2022-02-10:
This issue reproduces on Netgear R6220, Openwrt 19.07.3 r11063-85e04e9f46: 

```
The WAN is DHCP client, and ip address of WAN interface is 192.168.1.3.   
Before changing LAN interface IP address from 192.168.1.1 to 192.168.76.1 on LUCI, my PC connected to LAN can access Internet, and can access 192.168.1.1. 
After changing LAN interface IP address from 192.168.1.1 to 192.168.76.1 on LUCI, my PC connected to LAN can not access Internet, and can not access 192.168.1.1, and can not access 192.168.76.1. The DHCP server still give me address 192.168.1.100. 
Now I tried to manually set MyPC.ip to 192.168.76.66, and I still can not access 192.168.76.1. 
```

Now I hard-reset the router. 

This time, I modified /etc/config/network and run `/etc/init.d/network reload`. Before doing the modification, I can access 192.168.1.1 and the Internet. After doing the modification, I can access 192.168.76.1 and the Internet. 

Obviously, the behavior of LuCI is buggy.

Neustradamus, 2022-02-10:
@recolic: Yes, I can confirm the problem that I have already done previously.

Maybe it is needed to create a new ticket?

Or the @openwrt team (example @jow-) must reopen this ticket to solve it?

taylorsatula, 2023-01-13:
> FWIW doing it from command line works as expected.
> 
> Ssh into your device and assuming you want to move OpenWrt's LAN to the 192.168.2.x subnet:
> 
> ```
> root# uci set network.lan.ipaddr='192.168.1.2' ; uci commit network ; service network restart
> ```
> 
> Restart device just for good measure and try ssh-ing again using the new IP.

This worked for me. I popped open a ssh terminal and pasted it in. Unplugged the ethernet, counted to 5, plugged back in and navigated to the web interface at 192.168.1.2.

Weird. Whatever works..