openwrt/luci #2661: openwrt官方2019年4月4日源码编译后，运行出错。

tournet, 2019-04-07:
Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocks-libev/rules'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:165: Datatype error, bad token "cidr"
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/cbi.lua:165: in function 'compile_datatype'
	/usr/lib/lua/luci/cbi.lua:177: in function 'compile_datatype'
	/usr/lib/lua/luci/cbi.lua:189: in function 'verify_datatype'
	/usr/lib/lua/luci/cbi.lua:1493: in function 'validate'
	/usr/lib/lua/luci/cbi.lua:1414: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:248: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:1111: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:248: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:387: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:881: in function </usr/lib/lua/luci/dispatcher.lua:860>

tournet, 2019-04-07:
我发现问题了
/usr/lib/lua/luci/model/cbi/shadowsocks-libev/rules.lua里的
o.datatype = "or(ipaddr,cidr)"要修改成o.datatype = "or(ipaddr,string)"

hnyman, 2019-04-07:
English, please.

closed 2019-04-07 by hnyman