openwrt/luci #2665: Timezone Issue

fernandobohrer, 2019-04-07:
I've just figured out that changing the timezone settings through LuCI isn't working as expected.

Right now the GMT time is 18:10. Changing the timezone settings through the LuCI interface to GMT-3 gives me as result 21:10 [ instead of 15:10 ] and changing it to GMT+3 gives me 15:10 [ instead of 21:10 ]. The logic seems to be inverted.

At the terminal I can see the following:

Setting the timezone to GMT through LuCI:

```
root@TPLC60:~# date ; uci show | grep timezone
Sun Apr  7 18:13:54 GMT 2019
system.@system[0].timezone='GMT0'
```

Setting the timezone to GMT-3 through LuCI:

```
root@TPLC60:~# date ; uci show | grep timezone
Sun Apr  7 21:14:43 +03 2019
system.@system[0].timezone='<+03>-3'
```

Setting the timezone to GMT+3 through LuCI:

```
root@TPLC60:~# date ; uci show | grep timezone
Sun Apr  7 15:15:27 -03 2019
system.@system[0].timezone='<-03>3'
```

LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d) / OpenWrt 18.06.2 r7676-cddd7b4c77

jow-, 2019-04-07:
See https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html
Specifically this quote:
> The offset specifies the time value you must add to the local time to get a Coordinated Universal Time value.

So the zone offset is effectively subtracted from UTC, not the other way around.

closed 2019-04-07 by jow-

fernandobohrer, 2019-04-07:
@jow- thanks for the update on this.

Currently I'm in Porto Alegre. The city respects the same timezone as Sao Paulo. Setting the openwrt timezone parameter to 'America/Sao Paulo' produces the following output:

```
root@TPLC60:~# date ; uci show | grep timezone
Sun Apr  7 18:20:00 -03 2019
system.@system[0].timezone='<-03>3<-02>,M11.1.0/0,M2.3.0/0'
```
Considering that the Brazilian government announced earlier this week that for 2019/2020 we will not be going to have daylight saving time period here in Brazil, I wanted to configure openwrt to use the GMT-3. Setting the timezone parameter to 'GMT-3' produced the following output:

```
root@TPLC60:~# date ; uci show | grep timezone
Mon Apr  8 00:20:25 +03 2019
system.@system[0].timezone='<+03>-3'
```
That behaviour seemed wrong to me because GMT-3 refers to 3 hours behind GMT. It was not making sense why the clock was set forward instead of backwards.

With your response I found out that the 'Etc' before GMT parameter justifies the difference of the behaviour that I was expecting:

"The special area of "Etc" is used for some administrative zones, particularly for "Etc/UTC" which represents Coordinated Universal Time. In order to conform with the POSIX style, those zone names beginning with "Etc/GMT" have their sign reversed from the standard ISO 8601 convention. In the "Etc" area, zones west of GMT have a positive sign and those east have a negative sign in their name (e.g "Etc/GMT-14" is 14 hours ahead of GMT.)"

Source: [https://en.wikipedia.org/wiki/Tz_database#Area](https://en.wikipedia.org/wiki/Tz_database#Area)

With this in mind and setting the openwrt to 'Etc/GMT+3' I have now the output that I wanted:

```
root@TPLC60:~# date ; uci show | grep timezone
Sun Apr  7 18:25:05 -03 2019
system.@system[0].timezone='<-03>3'
```