openwrt/luci #2680: LuCi hangs

davidc502, 2019-04-22:
I've experienced this for myself but am also getting reports from users experiencing the same thing --- LuCi will hang randomly.  Which theme used does not matter. Nothing in the System or Kernel logs to indicate an issue.

The symptom is when trying to log in (but not always), or clicking on any of the links (but not always), LuCi will sit and spin like screenshot below.  Eventually, it will either log in or the page will load, but sometime can take minutes for that to happen.

Reported hardware: 
Linksys 3200acm
Linksys 32x
Linksys 1900acs

[image]

solidus1983, 2019-04-22:
I can also confirm this as well using David's build on WRT32x, Hangs not only during use after login but also at the login stage takes me a few spams of the button before it logs me in.

davidc502, 2019-04-23:
@solidus1983 
Are you also using dnscrypt-proxy version 2?

ErikBAndersen, 2019-04-23:
I also see these UI hangs.  Wrt3200acm, latest davidc builds.
I am not using dnscrypt-proxy version 2, but I am using the
older/stock dnscrypt-proxy v1.9 version.

jow-, 2019-04-23:
Please see if https://git.openwrt.org/f00a4ae6e0 helps.

solidus1983, 2019-04-23:
I changed two settings with dnscrypt on.

to
	option max_connections '50'
	option max_requests '10'
from
	option max_connections '500'
	option max_requests '1'

davidc502, 2019-04-23:
Thanks for testing... various users not using dnscrypt-proxy 2 have reported LuCi hanging, so this isn't dnscrypt-proxy related.

davidc502, 2019-04-23:
Will test the 2 HTTP(s) options this evening and ask others to test as well.

pmelange, 2019-04-24:
This is related to https://github.com/openwrt/luci/issues/2560

thanks @jow- for reverting to the previous settings

davidc502, 2019-04-25:
Closing this issue --

closed 2019-04-25 by davidc502

camelcamro, 2019-05-27:
hmm, still having same issue with newest trunk :(
should be that really fixed ?
pls reopen

jpdribbler, 2019-06-15:
same issue here :(

hnyman, 2019-06-15:
> same issue here :(

And you have modified your /etc/config/uhttpd as suggested above in 

> > Please see if https://git.openwrt.org/f00a4ae6e0 helps:
    
      option max_requests 3

jpdribbler, 2019-06-15:
> > same issue here :(
> 
> And you have modified your /etc/config/uhttpd as suggested above in
> 
> > > Please see if https://git.openwrt.org/f00a4ae6e0 helps:
> 
> ```
>   option max_requests 3
> ```

I just checked but it is already set to:

config uhttpd 'main'
	option max_requests '3'

facboy, 2019-06-22:
i have this problem too, mine is currently set to

```
option max_connections '50'
option max_requests '10'
```

dreary-ennui, 2019-06-23:
Having same problem, latest build, with connections set to 50 and requests set to 10.

teapethu, 2019-08-16:
I hit the same issue while the default config is already 3
`option max_requests '3'`

teapethu, 2019-08-19:
Fixed the issue by setting uhttpd configuration http_keepalive to '0'