openwrt/luci #2686: Basic Authentication (httpd.conf) 18.06.2

exp10r, 2019-05-01:
please help
ssh and LuCI login/password - root/root
in file /etc/config/uhttpd in section 'main'
option config '/etc/httpd.conf'
in file /etc/httpd.conf
/:admin:admin
After service restart uhttpd I log into the web interface. Asks to enter login in the window, I enter the admin admin and I get into the authentication LuCI. I enter the root root and get an error "Invalid username and/or password! Please try again."

in lede 17, it worked, what you need to do to make it work here?

feckert, 2019-05-02:
I had a similar problem.
Maybe add:

```
config login
	option username 'root'
	option password '$p$root'
	list read '*'
	list write '*'
```

to /etc/config/rpcd

exp10r, 2019-05-02:
it is, but it does not solve the problem.

jow-, 2019-05-02:
You need to change various places to modify the default username:

1) Add login to `/etc/config/rpcd`, crypt the password with `uhttpd -m` or `mkpasswd -5` on a Desktop. Example:
    ```
    uci add rpcd login
    uci set rpcd.@login[-1].username=admin
    uci set rpcd.@login[-1].password="$(uhttpd -m admin)"
    uci set rpcd.@login[-1].read="*"
    uci set rpcd.@login[-1].write="*"
    uci commit rpcd
    service rpcd restart
   ```

2) Add admin to the allowed login usernames

   ```
   vi /usr/lib/lua/luci/controller/admin/index.lua
   # Change `page.sysauth = "root"` to `page.sysauth = { "root", "admin" }`
   rm -r /tmp/luci-*
   ```

3) Add login to httpd.conf

    ```
    echo "/:admin:$(uhttpd -m admin)" > /etc/httpd.conf
    service uhttpd restart
    ```

closed 2019-05-02 by jow-

exp10r, 2019-05-02:
that's not exactly what you need.
I need double authorization: 
1 - Basic authentication via httpd.conf  
2 - through Luci.

jow-, 2019-05-02:
This is not supported anymore. When basic auth credentials are provided, the LuCI will use them for logging into the system and ignore the values from the login form.

exp10r, 2019-05-02:
does it depend on Luci?  can I install the old Luci?