openwrt/luci #2694: Need to add "vapid" option in wireless config.

khareanudeep, 2019-05-09:
we are facing issue of deleting Vap's :-
below are the steps by which we faced issue :-
1-suppose we have 3 vaps (3 ssid with 3 different mac-address)
2- and now i am going to delete 2nd ssid.
3-observation is 2 ssid deleted and 3 ssid's mac changed to 2'nd ssid mac.
this issue is resolved by the Qualcomm team by adding one para meter in wireless file
"option vapid 0" -> ath0
"option vapid 1" -> ath01
"option vapid 2" -> ath02

here my query is suppose i create vap from GUI so that **option vapid** by default not present i want at the time of creating vap it will add automatically "option vapid" according to creating vap.

Please help us on this as this is blocker issue.

khareanudeep, 2019-05-24:
hey any update on this ?

dibdot, 2019-05-24:
Just curios, which 'vapid' option are you talking about? Please post your wireless config (/etc/config/wireless) and provide the output of `ubus call system board.`

jow-, 2019-05-24:
This is unrelated to LuCI and upstream OpenWrt.  If you want to add custom option during interface creation, modify the `local wconf = { ... }` table of `/usr/lib/lua/luci//model/cbi/admin_network/wifi_add.lua`.

closed 2019-05-24 by jow-

khareanudeep, 2019-05-24:
@dibdot ,this is output of ubus call system board.
root@wips-test:/# ubus call system board
{
        "kernel": "3.14.77",
        "hostname": "wips-test",
        "system": "ARMv7 Processor rev 0 (v7l)",
        "model": "QCA IPQ8064\/AP161",
        "release": {
                "distribution": "OpenWrt",
                "version": "Chaos Calmer",
                "revision": "unknown",
                "codename": "chaos_calmer",
                "target": "ipq806x\/indoor",
                "description": "OpenWrt Chaos Calmer 15.05.1"
        }
}

khareanudeep, 2019-05-24:
hi @dibdot  **this is wireless config file:**

config wifi-device 'wifi0'           
        option type 'qcawifi'        
        option hwmode '11ac'          
        option htmode 'HT80'          
                option country 'IN'   
        option channel 'auto'         
        option txpower '23'           
                option disabled '0'   
                option country '5006' 
                                      
config wifi-device 'wifi1'            
        option type 'qcawifi'         
        option hwmode '11ng'          
        option htmode 'HT40+'         
                option htmode1 'HT40+'
        option channel 'auto'        
        option txpower '19'          
                option disabled '0'  
                option country 'IN'  
                                             
config wifi-iface                            
        option device 'wifi0'                
        option network 'lan'                 
        option mode 'ap'                     
        option ssid 'aaaaa'                  
        option encryption 'psk2+ccmp'        
        option key 'password'                
        option disabled '0'                  
        option hidden '0'                    
        option vhtmubfer '1'                 
        option rate_limit_enabled '0'        
        **option vapid '0'**                     
                                             
config wifi-iface                            
        option device 'wifi0'                
        option network 'lan'                 
        option mode 'ap'            
option ssid 'bbbbbb'         
        option encryption 'psk2+ccmp'
        option key 'password'        
        option disabled '0'          
        option hidden '0'                    
        option vhtmubfer '1'                 
        option rate_limit_enabled '0'        
        **option vapid '1'**                     
                                             
config wifi-iface                            
        option device 'wifi1'                
        option network 'lan'                 
        option mode 'ap'                     
        option ssid '11111'                
        option encryption 'psk2+ccmp'        
        option key 'password'                
        option disabled '0'                  
        option hidden '0'                    
        option rate_limit_enabled '0'        
        option disablecoext '1'              
        **option vapid '0'**

khareanudeep, 2019-05-24:
@jow- please tell me how to add custom option in  local wconf = { ... }.give me some deference

khareanudeep, 2019-05-30:
please reply