openwrt/luci #2696: Got exception when access to luci-app-shadowsocks-libev web page

1plus-1, 2019-05-12:
### Got below exception when access to luci-app-shadowsocks-libev page:
```
Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocks-libev/instances'.
The called action terminated with an exception:

...b/lua/luci/model/cbi/shadowsocks-libev/instances.lua:19: bad argument #1 to 'pairs' (table expected, got boolean)
stack traceback:
	[C]: in function 'pairs'
	...b/lua/luci/model/cbi/shadowsocks-libev/instances.lua:19: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:838: in function 'target'
	/usr/lib/lua/luci/dispatcher.lua:956: in function </usr/lib/lua/luci/dispatcher.lua:952>
```

### Following are system informaion
```
Model	Linksys WRT1900ACS
Architecture	ARMv7 Processor rev 1 (v7l)
Firmware Version	OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)
```

### luci app version is
```
luci-app-shadowsocks-libev - git-19.127.24963-84692e5-1
```

1plus-1, 2019-05-12:
Any suggestion on quick fix? Can I modify any file like "/usr/lib/lua/luci/model/cbi/shadowsocks-libev/instances.lua" on target to fix it?

hnyman, 2019-05-12:
> bad argument #1 to 'pairs' (table expected, got boolean)

The error is likely about missing/wrong config. The page logic expects contents for some dropdown list and there is nothing in the config file. Or something like that.  

> Any suggestion on quick fix?

Fix your shadowsocks config manually by editing the config file?
Revert to the default shadowsocks config file?

There have been lots of fixes/enhancements from @yousong in master, which have not been backported to 18.06. But it is really hard to say if any of those would fix your problem.

yousong, 2019-05-12:
Was `/etc/config/shadowsocks-libev` present?  If yes, please share the content of it with password and server name redacted.

1plus-1, 2019-05-13:
It's my fault. The configuration file was broken after editing by vi. I fixed it and the issue is gone.

Thanks for all the help!

closed 2019-05-13 by 1plus-1

t0nyt4ng, 2021-02-06:
> It's my fault. The configuration file was broken after editing by vi. I fixed it and the issue is gone.
> 
> Thanks for all the help!

I had the exactly the  same problem ,  but I did not change the configuration ,Please tell me how did you fix it ?