openwrt/luci #2698: tab "exec" on statistics (collectd) - not shown and not working since new trunk version

camelcamro, 2019-05-14:
on latest trunk it does not shown the tab "exec" on collectd 

https://192.168.1.5/cgi-bin/luci/admin/statistics/graph

if i try the link manualyl:
link: https://192.168.1.5/cgi-bin/luci/admin/statistics/graph/exec
gives the error:

/usr/lib/lua/luci/dispatcher.lua:393: Access Violation
The page at 'admin/statistics/graph/exec/' has no parent node so the access to this location has been denied.
This is a software bug, please report this message at https://github.com/openwrt/luci/issues
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:393: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

any ideas whats wrong ?

camelcamro, 2019-05-15:
ok, i checked now , which file is the wrong one ...
i overwrote the file: luci_statistics.lua
and then it was working again. so, there is the error anyhow ...
see /usr/lib/lua/luci/controller/luci_statistics
and attached the good(old) and the new(bad) ..

pls can you fix that anyhow ?

if wanted, i can add both files ...

[luci_statistics.lua.zip](https://github.com/openwrt/luci/files/3180378/luci_statistics.lua.zip)

hnyman, 2019-05-15:
One possible reason:
@feckert added new label definitions for most plugins with cd1e8cc6b94e in March, but apparently exec was not among the 27 plugins done at that time.   (Looking at the plugin list in staitistics, at least email and exec are missing from that commit.( But I haven't looked if labels for those are really needed.)

You might test if adding a similar label definition file for exec as for thee others would help.

feckert, 2019-05-15:
To get some value from system which could not get collected by a native collectd plugin, we normally use the exec plugin.
With my change could you still configure the exec plugin?
If you have written a definition file for your exec plugin and put the into the directory `applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions` you also have to add the item callback so it could be shown on the tab selection page.

camelcamro, 2019-05-26:
are you having an example for it ? sorry, but i don'T know what i should add more ...
the exec.lua demo file is included on the exec mod what i installed - i guess it is the same principle ...
or maybe the manpages are not up2date ?
what is here missing ?

my exec.lua file is looking like that:
/usr/lib/lua/luci/statistics/rrdtool/definitions/exec.lua
`
module("luci.statistics.rrdtool.definitions.exec", package.seeall)

function rrdargs(graph, plugin, plugin_instance)
    if "gsminfo" == plugin_instance then
        return {
            title = "%H: GSM Signal",
            vlabel = "dBI",
            number_format = "%5.1lf",
            y_min = "0",
            -- y_max = "100",
            -- alt_autoscale_max = true,
            autoscale_max = true,
            -- noarea = false,
            -- overlay = true,
            data = {
                types = { "gsm_signal" },
                sources = { gsm_signal = { "RSSI", "MODE", "RSRP", "RSCP", "RSRQ", "SINR", "ECIO" }
                -- sources = { gsm_signal = { "RSSI", "MODE", "RSRP", "RSCP", "RSRQ", "SINR", "ECIO" }
                },
                options = {
                    gsm_signal__MODE = { color = "ff0000", title = "Mode(4G=40 / 3G=30 / 2G=20)", overlay = true, noarea = false },  -- red
                    gsm_signal__RSSI = { color = "00bb00", title = "RSSI", overlay = true, noarea = false }, -- green
                    gsm_signal__RSRP = { color = "00ffff", title = "RSRP", overlay = true, noarea = true },  -- cyan  
                    gsm_signal__RSCP = { color = "0000ff", title = "RSCP", overlay = true, noarea = true },  -- blue  
                    gsm_signal__RSRQ = { color = "ff00ff", title = "RSRQ", overlay = true, noarea = true },  -- violet
                    gsm_signal__SINR = { color = "dd9900", title = "SINR", overlay = true, noarea = true },  -- orange 
                    gsm_signal__ECIO = { color = "000000", title = "ECIO", overlay = true, noarea = true }   -- black
                }
            }
        }
    end
 
    if "speedcheck" == plugin_instance then
        return {
            title = "%H: Download Speedcheck (SQM)",
            vlabel = "kbit/sec",
            number_format = "%5.3lf",
            -- y_min = "0",
            -- y_max = "1",
            -- alt_autoscale_max = true,
            autoscale_max = true,
            -- noarea = false,
            -- overlay = true,
            data = {
                types = { "SQM_speed" },
                sources = { SQM_speed = { "DOWNLOAD_SQM", "DOWNLOAD_REAL", "UPLOAD_SQM", "UPLOAD_REAL" }
                },
                options = {
                    SQM_speed__DOWNLOAD_SQM = { color = "00dd00", title = "Download Speed (SQM)", overlay = true, noarea = false, total = false }, -- green
                    SQM_speed__DOWNLOAD_REAL = { color = "00ffff", title = "Download Speed (REAL)", overlay = true, noarea = true, total = false }, -- cyan
                    SQM_speed__UPLOAD_SQM = { color = "bb0000", title = "Upload Speed (SQM)", overlay = true, noarea = false, total = false, flip = true }, -- red
                    SQM_speed__UPLOAD_REAL = { color = "dd9900", title = "Upload Speed (REAL)", overlay = true, noarea = true, total = false, flip = true } -- orange
                }
            }
        }
    end

   if "internet" == plugin_instance then
        return {
            title = "%H: Internet Speed (speedtest.net)",
            vlabel = "Mbit/sec",
            number_format = "%5.3lf",
[... 28 lines cut ...]
            -- noarea = false,
            -- overlay = true,
            data = {
                types = { "if_uptime" },
                sources = { if_uptime = { "MODEM", "MODEM_MW", "WAN", "WAN_MW", "VPNROUT03", "VPNROUT03_MW", "VPNROUT04", "VPNROUT04_MW", "VPNROUT05", "VPNROUT05_MW" }
                },
                options = {
                  if_uptime__MODEM = { color = "00dd00", title = "Uptime MODEM", overlay = true, noarea = false },                           -- #2 green
                  if_uptime__MODEM_MW = { color = "009900", title = "Online MODEM", overlay = true, noarea = true },           -- #2 green
                  if_uptime__WAN = { color = "007000", title = "Uptime WAN", overlay = true, noarea = true },                                -- #1 dark green
                  if_uptime__WAN_MW = { color = "003000", title = "Online WAN", overlay = true, noarea = true },               -- #1 dark green
                  if_uptime__VPNROUT03 = { color = "00ffff", title = "Uptime VPN_ROUT03", overlay = true, noarea = true },                   -- #5 cyan
                  if_uptime__VPNROUT03_MW = { color = "00bbbb", title = "Online VPN_ROUT03", overlay = true, noarea = true },  -- #5 cyan
                  if_uptime__VPNROUT04 = { color = "ff00ff", title = "Uptime VPN_ROUT04", overlay = true, noarea = true },                   -- #6 pink
                  if_uptime__VPNROUT04_MW = { color = "bb00bb", title = "Online VPN_ROUT04", overlay = true, noarea = true },  -- #6 pink
                  if_uptime__VPNROUT05 = { color = "990099", title = "Uptime VPN_ROUT05", overlay = true, noarea = true },                   -- #7 violet
                  if_uptime__VPNROUT05_MW = { color = "550055", title = "Online VPN_ROUT05", overlay = true, noarea = true }  -- #7 violet
                }
            }
        }
    end
    
end

`

camelcamro, 2019-05-26:
i guess we can close that ticket: (ort maybe you want to insert a demo package ?.
if wanted i can help to add such kind of ....

i had to add only the return through the translate stuff ...
/usr/lib/lua/luci/statistics/rrdtool/definitions/exec.lua
```

module("luci.statistics.rrdtool.definitions.exec", package.seeall)

+function item()
+	return luci.i18n.translate("Exec")
+end
+
function rrdargs(graph, plugin, plugin_instance)
    -- For $HOSTNAME/exec-foo-bar/temperature_baz-quux.rrd, plugin will be
    -- "exec" and plugin_instance will be "foo-bar".  I guess graph will be
    -- "baz-quux"?  We may also be ignoring a fourth argument, dtype.
    if "gsminfo" == plugin_instance then
 
```

feckert, 2019-05-26:
> +function item()
> +	return luci.i18n.translate("Exec")

You could add your individual translation. For example GSM info.

> +end

closed 2019-06-04 by feckert