openwrt/luci #2720: Error compiling lucihttp with openwrt 18.06.2

mportela, 2019-05-20:
Withou Luci this openwrt 18.06.2 compile without problem, but when i choose luci i have this:

```
make[3]: Entering directory '/code/openwrt/feeds/luci/contrib/package/lucihttp'
mkdir -p /code/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp
echo '2018-05-18-cb119ded' | cmp -s - /code/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/liblucihttp.version || echo '2018-05-18-cb119ded' > /code/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/liblucihttp.version
SHELL= flock /code/openwrt/tmp/.root-copy.flock -c 'cp -fpR /code/openwrt/build_dir/target-mips_24kc_musl/lucihttp-2018-05-18-cb119ded/.pkgdir/liblucihttp/. /code/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/'
touch /code/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp/.liblucihttp_installed
echo "liblucihttp" >> /code/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/lucihttp.default.install
rm -rf /code/openwrt/tmp/stage-lucihttp
mkdir -p /code/openwrt/tmp/stage-lucihttp/host /code/openwrt/staging_dir/target-mips_24kc_musl/packages /code/openwrt/staging_dir/host/packages
install -d -m0755 /code/openwrt/tmp/stage-lucihttp
cp -fpR /code/openwrt/build_dir/target-mips_24kc_musl/lucihttp-2018-05-18-cb119ded/ipkg-install/* /code/openwrt/tmp/stage-lucihttp/
find /code/openwrt/tmp/stage-lucihttp -name '*.la' | xargs -r rm -f;
if [ -f /code/openwrt/staging_dir/target-mips_24kc_musl/packages/lucihttp.list ]; then /code/openwrt/scripts/clean-package.sh "/code/openwrt/staging_dir/target-mips_24kc_musl/packages/lucihttp.list" "/code/openwrt/staging_dir/target-mips_24kc_musl"; fi
if [ -d /code/openwrt/tmp/stage-lucihttp ]; then (cd /code/openwrt/tmp/stage-lucihttp; find ./ > /code/openwrt/tmp/stage-lucihttp.files); 	SHELL= flock /code/openwrt/tmp/.staging-dir.flock -c ' mv /code/openwrt/tmp/stage-lucihttp.files /code/openwrt/staging_dir/target-mips_24kc_musl/packages/lucihttp.list && cp -fpR /code/openwrt/tmp/stage-lucihttp/* /code/openwrt/staging_dir/target-mips_24kc_musl/; '; fi
cp: not writing through dangling symlink '/code/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/liblucihttp.so'
Makefile:53: recipe for target '/code/openwrt/staging_dir/target-mips_24kc_musl/stamp/.lucihttp_installed' failed
make[3]: *** [/code/openwrt/staging_dir/target-mips_24kc_musl/stamp/.lucihttp_installed] Error 1
make[3]: Leaving directory '/code/openwrt/feeds/luci/contrib/package/lucihttp'
time: package/feeds/luci/lucihttp/compile#0.05#0.03#0.58
package/Makefile:107: recipe for target 'package/feeds/luci/lucihttp/compile' failed
make[2]: *** [package/feeds/luci/lucihttp/compile] Error 2
make[2]: Leaving directory '/code/openwrt'
package/Makefile:103: recipe for target '/code/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/code/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/code/openwrt'
/code/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

can some one help with this?

jow-, 2019-05-20:
Retry after a `make package/lucihttp/clean`

mportela, 2019-05-20:
works! I was building and testing diferent openwrt versions !

closed 2019-05-20 by mportela