openwrt/luci #2722: Wifi Schedule Disable ssid

e2thenegpii, 2019-05-21:
It would be nice to disable wifi according to a schedule by wireless interface.

dibdot, 2019-05-21:
Check luci-app-wifischedule and/or this forum thread https://forum.openwrt.org/t/scheduling-on-off-wifi/3385

closed 2019-05-21 by dibdot

e2thenegpii, 2019-05-21:
Sure, just not a way to do it via Luci.

feckert, 2019-05-21:
@e2thenegpii there is also a luci front-end available.
https://github.com/openwrt/luci/tree/master/applications/luci-app-wifischedule

e2thenegpii, 2019-05-21:
Yes, this is what you get for submitting a ticket late at night, I didn't explain myself very well.

My intent of this ticket is that I didn't see a way to disable a specific wireless interface via luci-app-wifischedule.

User Story:
On my router I have two wifi interfaces we'll call them "kids" and "adults"  I would like to disable the kids wireless between 10pm and 7am, but leave the adults wifi available at all times.

Now I know there are other (arguably better) ways to solve this problem via dnsmasq tagging and iptables, I just wanted a quick brute force solution to disable a specific ssid via Luci.