openwrt/luci #2723: Unable to upgrade the firmware for Qualcomm Atheros AP148 referenceboard

Deepthidill, 2019-05-23:
Hi,
I am using senao Accesspoint(model:OAP8250AG(Qualcomm Atheros AP148 reference board )) with kernel version 3.4.103 and the current firmware version is QSDK Standard IPQ806X.LN / LuCI 0.11.1 Release (0.11.1) .it is 802.11ac supported device.to increase my router speed, iam going to upgrading the firmware through UI.(open the router page 192.168.1.1 navigate to system->backup/flashfirmware->flas new firmware image->upload the file.
(downloaded the firmware's from       http://archive.openwrt.org/releases/18.06.1/targets/ipq806x/generic/

qcom_ipq8064-ap148-fit-uImage.itb
qcom_ipq8064-ap148-legacy-squashfs-nand-factory.bin
qcom_ipq8064-ap148-legacy-squashfs-nand-sysupgrade.bin
qcom_ipq8064-ap148-squashfs-nand-factory.bin
qcom_ipq8064-ap148-squashfs-nand-sysupgrade.bin

observations:
-----------------
i am unable to upgrade the firmware due to the uploaded image file does not contain a supported format.

please help me to upgrade the firmware version which is support for 802.11ac wave2.

feckert, 2019-05-23:
This is not a LuCI problem. Please have a look on the Wiki-Page [1] or Forum-Page [2].
I would also suggest to do an upgrade over the shell/ssh session (The running system it very old)

[1] https://openwrt.org/
[2] https://forum.openwrt.org/

closed 2019-05-23 by feckert

Deepthidill, 2019-05-24:
Hi Feckert,
I have tried through ssh, but i am unable to update the opkg to download the file.please suggest me,did i missed any steps to update .

[image]

dibdot, 2019-05-24:
You can't use 'opkg' to download firmware images - just wget image files directly to the /tmp directory of your router and flash from there ... anyway, please join the forum for further questions in this regard.