openwrt/luci #2726: luci-app-transmission: Option `prefetch_enabled` is set as integer instead of boolean

salim-b, 2019-05-24:
I just noticed that the transmission option `prefetch_enabled` is not correctly reflected in the LuCI app (unchecked though I've set it to `true` in `/etc/config/transmission`).

When I check this option in LuCI, I can see that the configuration would be changed as follows:

```
uci set transmission.cfg015f8f.prefetch_enabled='1'
```

I think this is wrong because this option should be a boolean, i.e. the config change should read:

```
uci set transmission.cfg015f8f.prefetch_enabled='true'
```

I'm using stable OpenWrt 18.06.2 on a TP-Link Archer C2600 if that matters.

dibdot, 2019-05-25:
Boolean options may have one of the values '0', 'no', 'off', 'false' or 'disabled' to specify a false value or '1' , 'yes', 'on', 'true' or 'enabled' to specify a true value

closed 2019-05-25 by dibdot

salim-b, 2019-05-26:
> Boolean options may have one of the values '0', 'no', 'off', 'false' or 'disabled' to specify a false value or '1' , 'yes', 'on', 'true' or 'enabled' to specify a true value

Ok, thanks for the clarification!

But why is the `prefetch_enabled` option not correctly reflected in the LuCI app, i.e. unchecked though I've set it to `true` in `/etc/config/transmission`?

dibdot, 2019-05-26:
The used 'flag' object in this app describing an option with two possible values in a section of a UCI File, default values are '0' and '1' ... it's all OK, just use these values.

salim-b, 2019-05-26:
> The used 'flag' object in this app describing an option with two possible values in a section of a UCI File, default values are '0' and '1'

I see, so this app doesn't actually implement [all the possibilities of boolean options you've described](https://github.com/openwrt/luci/issues/2726#issuecomment-495864096)... is this a limitation of LuCI in general?

> it's all OK, just use these values.

Well, the [default config of OpenWrt's transmission package](https://github.com/openwrt/packages/blob/master/net/transmission/files/transmission.config) sets `option prefetch_enabled true` (and not `1`), so the current situation seems somewhat less than perfect.