openwrt/luci #2728: Led not configurable through luci after upgrading to 18

stephengeorgewest, 2019-05-26:
Related to #971 
I just  updated to OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d) and now I only get one port as an option.
[image]

https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/luasrc/model/cbi/admin_system/leds.lua#L151 says to use "/sys/bus/usb/devices/usb*" if present, but that is the usb port of the built in hub.
[image]

In my case I think it should be listing 1-1.1 as port 1 and 1-1.2 as port 2. I'm not sure how to fix it without breaking everyone else.

I guess not many other systems have a useless hub permanently attached to the root hub. Maybe I didn't have this extra hub before (I don't remember doing a lsusb), and openwrt/openwrt issue, or kernal updates...
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 051d:0002 American Power Conversion Uninterruptible Power
Bus 001 Device 004: ID 051d:0002 American Power Conversion Uninterruptible Power

stephengeorgewest, 2019-05-27:
I read that bit wrong. That issue was just to read the values from the configuration. I think the issue is in one or both of the "for p in nixio.fs.glob". still trying stuff.

stephengeorgewest, 2019-05-27:
Model	TP-Link TL-WDR3600 v1
Architecture	Atheros AR9344 rev 2
Firmware Version	OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d)
Kernel Version	4.9.152

Probably should have put that in at the top.

If i put <code>usbport:value("usb1-port2", "Hub 1, Port 2*")<code> after the <code>for p in nixio.fs.glob("/sys/bus/usb/devices/*/usb[0-9]*-port[0-9]*") do  </code> .. <code>end</code> block I get:
[image]

[image]
I don't know if that helps anything. I noticed that the led's don't even turn off when unplugged. That's probably a different problem.

jow-, 2019-05-28:
Can you please paste the complete output of `find /sys/bus/usb/devices/` ?

stephengeorgewest, 2019-05-30:
root@TP-Link:~# find /sys/bus/usb/devices/
/sys/bus/usb/devices/
/sys/bus/usb/devices/1-1.2:1.0
/sys/bus/usb/devices/usb1
/sys/bus/usb/devices/1-1.1
/sys/bus/usb/devices/1-1
/sys/bus/usb/devices/1-0:1.0
/sys/bus/usb/devices/1-1.1:1.0
/sys/bus/usb/devices/1-1.2
/sys/bus/usb/devices/1-1:1.0
root@TP-Link:~#

stephengeorgewest, 2019-08-12:
Due to unrelated issue, I reflashed with the latest firmware, and wiped the settings. I noticed that the leds light up when plugged.
```
system.led_usb1=led
system.led_usb1.name='USB1'
system.led_usb1.sysfs='tp-link:green:usb1'
system.led_usb1.trigger='usbdev'
system.led_usb1.interval='50'
system.led_usb1.dev='1-1.1'
system.led_usb2=led
system.led_usb2.name='USB2'
system.led_usb2.sysfs='tp-link:green:usb2'
system.led_usb2.trigger='usbdev'
system.led_usb2.interval='50'
system.led_usb2.dev='1-1.2'
system.led_wlan2g=led
system.led_wlan2g.name='WLAN2G'
system.led_wlan2g.sysfs='tp-link:blue:wlan2g'
system.led_wlan2g.trigger='phy0tpt'

```
I noticed that .trigger is 'usbdev' but that isn't an option in the trigger list (but [usbport] is):

```
root@TP-Link:~# cat /sys/class/leds/tp-link:green\:usb1/trigger
none switch0 timer default-on netdev [usbport] phy0rx phy0tx phy0assoc phy0radio phy0tpt phy1rx phy1tx phy1assoc phy1radio phy1tpt
```
`\etc\init.d\led` says "usbdev" is old, and should be "usbport"
I noticed that a `/sys/bus/usb/devices/*/*-port[0-9]*` showed the available ports. and the format matches `/sys/class/leds/*/ports/`

I modified `\usr\lib\lua\luci\model\cbi\admin_system\leds.lua` to list all the value.

stephengeorgewest, 2019-08-12:
[image]

Neustradamus, 2022-08-10:
@stephengeorgewest: Have you progressed on this PR?

feckert, 2022-08-11:
Openwrt 18.06 is EOL!
So I will close them

closed 2022-08-11 by feckert