openwrt/luci #2729: static ipv6 address setting gets wiped upon applying iface settings in LuCI

n8v8R, 2019-05-27:
18.06 trunk
___
Whilst LuCI is lacking an option to set/specify a static ipv6 address for `odhcp` it is still possible to do so manually in /etc/config/network with

`list ip6addr 'ipv6 address'`

However, manipulating the iface settings later on via LuCI and pressing apply the aforementioned setting simply gets wiped from /etc/config/network

feckert, 2019-05-27:
stupid question:
If you set the proto to static is it then possible to set the ip with LuCI?

n8v8R, 2019-05-27:
no, it does not matter/make a change. LuCI does not have a provision at all for setting/assigning a static ipv6 address. But that does not mean that is should wipe `list ip6addr`  from /etc/config/network, being a viable `odhpd` setting.

jow-, 2019-05-28:
I do not understand the bug report. Can you please post a complete /etc/config/network that gets clobbered by LuCI upon save?

What exactly do you mean with `specify a static ipv6 address for an iface via odhcp` ?

n8v8R, 2019-05-28:
My bad - it is not an `odhcpd` setting indeed but rather a `netifd` setting apparently

https://openwrt.org/docs/guide-user/network/ipv6/start
___
Similar to assigning an ipv4 address to an iface `ip6addr` assigns an ipv6 address to a particular interface.

LuCI does not provision for it and it seems that upon saving changed iface settings it validates the iface settings and because of not recognising `ip6addr` as a viable iface settings it wipes it from the iface config.

In this example for demo purpose juste changed the last digit of the ipv4 address via LuCI and pressed _save_, resulting in

[image]
___
Hope that clarifies

jow-, 2019-05-28:
Please provide the complete /etc/config/network file.

n8v8R, 2019-05-28:
```
config interface 'mgt'
	option type 'bridge'
	option bridge_empty '1'
	option proto 'static'
	option ipaddr '192.168.112.12'
	option netmask '255.255.255.0'
	list ip6addr 'fd30:d64c:1eed:4c3a::12'
	option ip6assign '64'
	option ip6hint '4c3a'
	list ifname 'lan3'
	list ifname 'dummy0'
```

jow-, 2019-05-28:
The `ip6assign` and `ip6addr` options are mutually exclusive (in LuCI) - that is why `ip6addr` gets removed upon save. Will clarify if they`re exclusive in netifd as well.

n8v8R, 2019-05-28:
I see, thanks for the clarification.

I do not see though why they should be mutually exclusive, at least to my understanding `ip6assign` pertains to the public prefix delegated from upstream whilst `ip6addr` does not?

jow-, 2019-05-28:
They were made exclusive in LuCI because for the vast majority of use cases it makes no sense to both use ip6assign and a static ip6addr setting. It would simply confuse users.

n8v8R, 2019-05-28:
It confused me that `LuCI` removes a setting that works for `netifd` and thus I opened this as a bug report, since I also could not find said exclusivity documented.

If it is a feature however then please feel at liberty to close the issue. I am aware of what is happening with LuCI and handle the matter accordingly (editing /etc/config/network via ssh instead).

n8v8R, 2019-06-07:
closing since understand this not being a(n) bug/issue.

closed 2019-06-07 by n8v8R