openwrt/luci #2739: i18n-scan.pl finds commented out code

karlp, 2019-05-31:
Not _reallllly_ a big deal per se, but it finds and generates msgids for commented out code.
```translate-fail.lua
--x = translate("this is commented out and shouldn't be seen")
```
output
```
$ ./build/i18n-scan.pl translate-fail.lua 
msgid ""
msgstr "Content-Type: text/plain; charset=UTF-8"

#: translate-fail.lua:1
msgid "this is commented out and shouldn't be seen"
msgstr ""
```

jow-, 2019-05-31:
Fixing this would be very complex and would require implementing complete Lua, ECMAScript and HTML syntax support in `i18n-scan.pl`, due to that I'll close this without a fix since I consider the impact of this to be neglectable.

closed 2019-05-31 by jow-

karlp, 2019-05-31:
that's fine.  The work in progress I have for the other one, https://github.com/openwrt/luci/issues/2738 woudl automatically handle this too, so let's leave thsi closed