openwrt/luci #2740: Network interface name longer than 12 chars and include it in firewall zone brick router

rcalpha, 2019-06-01:
If you create a new Network interface with a name longer than 12 characters, and then you include this in a firewall zone (new or existing), then apply the change, it will brick the router.

To recover, enter fail safe mode, delete this interface and firewall zone

jow-, 2019-06-01:
Sounds like something that should be fixed in OpenWrt core.

hnyman, 2019-06-01:
Reference to bug report & discussion from year 2015 in the old OpenWrt bug tracker:
https://dev.archive.openwrt.org/ticket/20380.html

(limitations from core Linux, which we do not enforce in Openwrt config)

n8v8R, 2019-06-02:
Core limit is 16 characters though and not 12

jow-, 2019-06-02:
Probably depends on the protocol and thus on the interface prefix length `br-...`, `6in4-...`, `pppoe-...` etc.

n8v8R, 2019-06-02:
right, those prefixes.

Oh, then there is anyway the hint in LuCI 

> Maximum length of the name is 15 characters including the automatic protocol/bridge prefix (br-, 6in4-, pppoe- etc.)

hnyman, 2019-06-02:
> Oh, then there is anyway the hint in LuCI

Yep.
I added that as a quick&dirty fix/advice after discussion in that ticket20380 and LuCI issue #507 with https://github.com/openwrt/luci/commit/b1217c88c3566c1bd726bce9203da591af564bcf 

I also added in 2015 the enforcement of only 11 chars for zone name in LuCI with https://github.com/openwrt/luci/commit/87b6bb0dafbbe46dc7669356ac3db24aa975c4b7 , but that is naturally not enforced on the main OpenWrt uci side, so you can edit the config file manually and have a longer name anyway...

As the real allowed interface name length depends on the protocol (which can be changed), it would be hard to set a 100% proof limit unless we shorten the allowed name quite much.

Alex85pro, 2019-06-04:
> ticket20380 and LuCI issue #507 with b1217c8
> LuCI with 87b6bb0 , but that is naturally not enforced on the main

closed 2025-02-27 by systemcrash