openwrt/luci #2745: PPPoE Debug Not Working with luci-app-mwan3

cpkern, 2019-06-03:
Currently running OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152) on Linksys WRT3200ACM with Kernel 4.14.63. Before installing luci-app-mwan3, the debug option for PPPD was working with all LCP EchoReq and LCP EchoRep statements appearing within the system logs. e.i. 
    ...    
sent [LCP EchoReq id=0x3e magic=0x5f5fe3cf]
rcvd [LCP EchoRep id=0x3e magic=0x3bdda1dc]
sent [LCP EchoReq id=0x3f magic=0x5f5fe3cf]
rcvd [LCP EchoRep id=0x3f magic=0x3bdda1dc]
    ...   
Since installing MWAN3 for a dual-WAN configuration, these statements no longer appear within the logs. The "dump" option is the only feature for PPPD that seems to be working. Upon reconnecting, the "dump" of the PPPoE options/commandline appears within the logs. Returning the router to its previous "one-WAN" configuration with PPPoE, and the statements return. Resetting the router to the new configuration, and the statements disappear once again. The router has its WAN port set for an ISP with PPPoE in both configurations, and the secondary WAN is set to employ the VLAN4 port, and the ISP to that port has my router using a static-IP or I can connect to the ISP as a DHCP client. Dual-WAN configuration is working well, but I'd like to debug the PPPoE side as this connection frequently resets when the ISP fails to respond to my router's LCP EchoReq's. It can take several minutes to reconnect on this PPPoE side, sometimes, it can take even longer. An example of the debugging to the logs when the router is set to employ only one WAN (original configuration prior to installing the MWAN3 package(s)) follows:

    ...   
sent [LCP EchoReq id=0x9 magic=0xa1e03747]
rcvd [LCP EchoRep id=0x9 magic=0x1653623c]
sent [LCP EchoReq id=0xa magic=0xa1e03747]
sent [LCP EchoReq id=0xb magic=0xa1e03747]
sent [LCP EchoReq id=0xc magic=0xa1e03747]
sent [LCP EchoReq id=0xd magic=0xa1e03747]
No response to 4 echo-requests
Serial link appears to be disconnected.
Connect time 52.0 minutes.
Sent 8378215 bytes, received 237222100 bytes.
Script /lib/netifd/ppp-down started (pid 7594)
sent [LCP TermReq id=0x3 "Peer not responding"]
Script /lib/netifd/ppp-down finished (pid 7594), status = 0x1
sent [LCP TermReq id=0x4 "Peer not responding"]
Connection terminated.
Send PPPOE Discovery V1T1 PADT session 0x202 length 0
 dst 00:00:00:00:00:00  src ff:ff:ff:ff:ff:ff
Sent PADT
Modem hangup
    ...    
          Minutes later...
    ....    
Send PPPOE Discovery V1T1 PADI session 0x0 length 12
 dst ff:ff:ff:ff:ff:ff  src 00:00:00:00:00:00
Recv PPPOE Discovery V1T1 PADO session 0x0 length 24
 dst 00:00:00:00:00:00  src ff:ff:ff:ff:ff:ff
Send PPPOE Discovery V1T1 PADR session 0x0 length 12
dst 00:00:00:00:00:00  src ff:ff:ff:ff:ff:ff
Recv PPPOE Discovery V1T1 PADS session 0x203 length 12
dst 00:00:00:00:00:00  src ff:ff:ff:ff:ff:ff
PPP session is 515
Connected to 00:00:00:00:00:00 via interface eth0.2
using channel 2
Using interface pppoe-wan
Connect: pppoe-wan <--> eth0.2
sent [LCP ConfReq id=0x5 <mru 1500> <magic 0x2d8b6f63>]
rcvd [LCP ConfReq id=0x1 <auth chap MD5> <mru 1480> <magic 0x5b6589e>]
sent [LCP ConfAck id=0x1 <auth chap MD5> <mru 1480> <magic 0x5b6589e>]
rcvd [LCP ConfAck id=0x5 <mru 1464> <magic 0x2d8b6f63>]
sent [LCP EchoReq id=0x0 magic=0x2d8b6f63]
rcvd [LCP EchoRep id=0x0 magic=0x5b6589e]
rcvd [CHAP Challenge id=0x1 <XXX>, name = "XXX"]
sent [CHAP Response id=0x1 <XXX>, name = "XXX"]
rcvd [CHAP Success id=0x1 "Welcome."]
CHAP authentication succeeded: Welcome.
CHAP authentication succeeded
peer from calling number 00:00:00:00:00:00 authorized
sent [IPCP ConfReq id=0x3 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
rcvd [IPCP ConfReq id=0x1 <addr 10.0.0.1>]
sent [IPCP ConfAck id=0x1 <addr 10.0.0.1>]
rcvd [proto=0x8281] 01 01 00 04
Unsupported protocol 0x8281 received
sent [LCP ProtRej id=0x6 82 81 01 01 00 04]
rcvd [IPCP ConfNak id=0x3 <addr X.X.X.X > <ms-dns1 X.X.X.X> <ms-dns2 X.X.X.X>]
sent [IPCP ConfReq id=0x4 <addr X.X.X.X> <ms-dns1 X.X.X.X> <ms-dns2 X.X.X.X>]
rcvd [IPCP ConfAck id=0x4 <addr X.X.X.X> <ms-dns1 X.X.X.X> <ms-dns2 X.X.X.X>]
local  IP address X.X.X.X
remote IP address 10.0.0.1
primary   DNS address X.X.X.X
secondary DNS address X.X.X.X
Script /lib/netifd/ppp-up started (pid 8341)
Script /lib/netifd/ppp-up finished (pid 8341), status = 0x1
sent [LCP EchoReq id=0x1 magic=0x2d8b6f63]
rcvd [LCP EchoRep id=0x1 magic=0x5b6589e]
sent [LCP EchoReq id=0x2 magic=0x2d8b6f63]
rcvd [LCP EchoRep id=0x2 magic=0x5b6589e]
    ...    
          Dump:
    ...    
Plugin rp-pppoe.so loaded.
RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
pppd options in effect:
debug debug             # (from command line)
nodetach                # (from command line)
persist         # (from command line)
logfile /var/log/pppdDebug.log          # (from /etc/ppp/options)
ifname pppoe-wan                # (from command line)
maxfail 0               # (from command line)
dump            # (from /etc/ppp/options)
ip-up-script /lib/netifd/ppp-up         # (from command line)
ip-down-script /lib/netifd/ppp-down             # (from command line)
ipv6-up-script /lib/netifd/ppp6-up              # (from command line)
ipv6-down-script /lib/netifd/ppp-down           # (from command line)
plugin rp-pppoe.so              # (from command line)
user XXX              # (from command line)
password ??????         # (from command line)
eth0.2          # (from command line)
noaccomp                # (from /etc/ppp/options)
mru 1500                # (from command line)
mtu 1500                # (from command line)
nopcomp         # (from /etc/ppp/options)
passive         # (from /etc/ppp/options)
lcp-echo-failure 3              # (from /etc/ppp/options)
lcp-echo-interval 4             # (from /etc/ppp/options)
ipparam wan             # (from command line)
noipdefault             # (from /etc/ppp/options)
nodefaultroute          # (from command line)
usepeerdns              # (from command line)
Interface eth0.2 has MTU of 1472 -- should be at least 1500.
    ...    

[image]

feckert, 2019-06-28:
Sorry I have no clue why this should not work without dig into it this issue.
I will have a look If I have more time sorry :-(

feckert, 2019-09-13:
This is not related to luci-app-mwan3.
Please reopen this Ticket in the package feed and ping me with the @ option because this is an issue for the mwan3 package.
Thanks

closed 2019-09-13 by feckert