openwrt/luci #2752: Firewall Traffic rules - 'on page' search not working

arcadejust, 2019-06-05:
Hi, While browsing Firewall Traffic rules using chrome search (Ctrl+F) does not produce any results for Traffic rules names - luci displays rule name as: div data-title. Chrome cannot select or search this property, when you have many rules it is impossible to navigate to designated rule to enable/disable it. Is there any configuration for this? Could it be done like Match or Action column (searching works there)?

jow-, 2019-06-05:
This is due to the fact that row captions are rendered via CSS to allow for responsive mobile styling. A fix for this will be complicated so don't expect any solution for this in the short term.

arcadejust, 2019-06-05:
Thank you. Do you know if it's possible to load an older version of this module without that functionality?

chuck-r, 2020-01-22:
Great job, guys, on the UI changes between 18.06 and 19.07. It all looks great! I'm the type of person that navigates my computer mainly with a keyboard, only touching the mouse when absolutely necessary. So, if I'm trying to find something on a web page (especially if I know the phrase or part of it) I always use Ctrl-F. Naturally, this is an issue that has bugged me since I installed OpenWRT just after the re-merge from LEDE.

I have about 40 firewall rules as of right now and while I'm pretty familiar with the names, I don't always remember the exact ports or MAC addresses related to the rules immediately. This leaves me having to scroll down the page for a bit, trying not to lose my place as my eyes track the screen. For me, this can be a bit difficult sometimes. Honestly, it would be easier if I could Ctrl-F and get what I came there for.

I did a little experimenting in Firefox and Chrome's dev consoles and this is what I came up with:

[image]

The changes made are shown in red. I disabled the ::before rule and added a <div><strong></strong></div> to the beginning of each row. My observation is that (at least as far as Firefox and Chrome are concerned) there is only a very minor difference between the two versions -- we're talking 1-2px padding/margin in the table column headers. Actually, the column header that ::before renders appears to actually be misaligned when compared to my version; the baselines don't line up if I'm seeing it correctly. As far as responsiveness on windows resizing and such, there is no observable difference between them.

Unfortunately, I am currently unable to test it in Chrome for Android, Firefox for Android, Safari, Opera, IE?, Edge or other browsers, so I'm unsure how that would differ. And, unfortunately, I haven't the time right now to modify the code to render my version to show it running live. But, please let me know what you think or what problems you might/do observe with this layout.

chuck-r, 2020-01-23:
Here's a side-by-side (new version on the left, old version on the right). The same changes were applied to the page as before. This does look a little different as it's missing some style but doesn't render oddly on mobile. I even went as far as making a rule name overflow and the result was the same on both versions.

The screenshot is with a small display size, but I also tried it with the largest display size with the same results.

[image]

adelton, 2025-01-10:
I've now checked with Firefox133.0.3 on Fedora 41 and Firefox 133 on Android that Ctrl+F / Find in page can find traffic rule names fine, on OpenWrt 23.05.5.

Is this still and problem with recent Chrome and recent OpenWrt?

closed 2025-01-10 by systemcrash