openwrt/luci #2771: [OpenWrt 18.06] LuCI can install packages even those which are in conflict

BKPepe, 2019-06-15:
In my case, file `/sbin/ntpd` is provided by busybox. When I want to install _ntpd_ package with **opkg** it ends up with:

```
root@router_name:~# opkg install ntpd
Installing ntpd (4.2.8p11-1.0) to root...
Downloading ****/ntpd_4.2.8p11-1_aarch64_cortex-a53.ipk
Installing ntp-utils (4.2.8p11-1.0) to root...
Downloading ****/ntp-utils_4.2.8p11-1_aarch64_cortex-a53.ipk
Configuring ntp-utils.
Collected errors:
 * check_data_file_clashes: Package ntpd wants to install file /sbin/ntpd
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package ntpd.
```

When I go to LuCI and in System → Software, I want to download and install package **ntpd** and once I press OK, then the LuCI doesn't have any problem with conflict.
```
Installing ntpd (4.2.8p11-1.0) to root...
Downloading ****/ntpd_4.2.8p11-1_aarch64_cortex-a53.ipk
Configuring ntpd.
```

Here is the output of opkg list-installed:
```
root@router_name:~# opkg list-installed | grep ntpd
ntpd - 4.2.8p11-1.0
```

And indeed right now `/sbin/ntpd` is provided by that package. You can see even the difference in help.

hnyman, 2019-06-15:
LuCI seems to take a brute-force approach and uses "--force-overwrite":

https://github.com/openwrt/luci/blob/4bbe32548c407ec6ae104934b3f1fe53539de9f6/applications/luci-app-opkg/luasrc/controller/opkg.lua#L29

https://github.com/openwrt/luci/blob/4bbe32548c407ec6ae104934b3f1fe53539de9f6/libs/luci-lib-ipkg/luasrc/model/ipkg.lua#L15

jow-, 2019-06-15:
This is intentional and will be kept for the time being until the package universe is mature enough to handle alternatives, conflicts and replacements cleanly.

closed 2019-06-15 by jow-

n8v8R, 2019-06-15:
This two way approach is somewhat incomprehensible, particularly when assuming the less nerdy using the GUI and being forcefully exposed to potentially conflicting packages, that may cause havoc and subsequent support tickets, whilst the potentially more advanced on the cli are protected from such conflicts?

jow-, 2019-06-15:
So far I've never had any support case related to this but plenty related to not instalable packages.

Cynerd, 2019-06-18:
@jow- I don't think that that is true. I am not saying that you personally seen some issue for exactly this I am just saying that this can cause other issues and hides underlying  problem.

The issue I see this can cause is when user tries to upgrade some package. In general upgrades are not supported in OpenWRT because they are broken. But they are broken because of decisions like these! So you can't just say: nobody complains about not be able to install packages they want. It is more about what is right and this is just wrong. You are breaking system without telling it to user. User with good intentions installs some package and you, instead of informing him that it is something that is not easily done, just do it and effectively break system to him.

Note that this is not just some inconvenience but rather pretty significant bug. You are forcing opkg to install package that should effectively be not there. Opkg, because it is not clever enough, looses track of which package owns given file and it is effectively random. If there is for example light and heavy variant of same library and you install them and later install some application able to run only with heavy version then such application can be broken. In such case there is nothing wrong with opkg nor with packages it installed. The cause is that you are setting up system to broken state and you are telling us here that it is ok solution?

I think that this patch should be accepted and users should be notified about that. This is same level of error such as collision between packages caused by `Conflicts:` (which btw is also ignored?). Solution should either happen automatically or it should be up to the user. Any solution of "let's ignore it" is just wrong.

jow-, 2019-06-18:
> The issue I see this can cause is when user tries to upgrade some package.

No. Package upgrades are not broken due to the use of `--force-overwrite`, maybe you confuse it with `--force-depends`.

> In general upgrades are not supported in OpenWRT because they are broken. But they are broken because of decisions like these! 

Upgrades are unsupported due to deficiencies in the package repo (lack of library ABI tracking. lack of proper maintenance of Supersedes/Replaces/Conflicts tags in package metadata, lack of central coordination for package transitions). The decision to use `--force-overwrite` is completely unrelated to that.

> So you can't just say: nobody complains about not be able to install packages they want. It is more about what is right and this is just wrong. You are breaking system without telling it to user. [...]

I still think that you somehow confuse `--force-depends` with `--force-overwrite` here. Packages that provide the very same files without having a corresponding `Conflicts` tag in their control file are usually equivalent providers of the same virtual package. LuCI historically allowed this usecase because atomically swapping one discrete virtual package provider for another is very hard to do and requires force flags even on the cli. Supplying `--force-overwrite` allows to install one provider (e.g. `libustream-mbedtls`) over another (e.g. `libustream-openssl`) and a subsequent removal of the other provider. Note that this has nothing to do with ignoring dependencies or ignoring `Conflicts: ` declarations in the control files.

> Note that this is not just some inconvenience but rather pretty significant bug. You are forcing opkg to install package that should effectively be not there. Opkg, because it is not clever enough, looses track of which package owns given file and it is effectively random.

It is not random. The package getting installed will assume ownership of the conflicting files. If the partially overwritten package is removed later, the conflicting files are retained.

> If there is for example light and heavy variant of same library and you install them and later install some application able to run only with heavy version then such application can be broken.

This cannot happen unless that application has a wrong dependency specification or a wrong `Conflicts` tag in its control file. If the application in question only works with `libfoo-heavy` then it must depend on `libfoo-heavy` and not `libfoo` provided by either `libfoo-light` or `libfoo-heavy`.

> In such case there is nothing wrong with opkg nor with packages it installed. The cause is that you are setting up system to broken state and you are telling us here that it is ok solution?

Again. Neither do we ignore `Depends: `, not `Conflicts: `. Neither of these cases you outlined are affected by `--force-overwrite`.

> I think that this patch should be accepted and users should be notified about that. This is same level of error such as collision between packages caused by Conflicts: (which btw is also ignored?). 

It is not ignored.

> Solution should either happen automatically or it should be up to the user. Any solution of "let's ignore it" is just wrong.

There are no such solutions available in opkg. There is no atomic `opkg replace ...` operation available which would cleanly address this case.

Cynerd, 2019-06-19:
> No. Package upgrades are not broken due to the use of --force-overwrite, maybe you confuse it with --force-depends.

I am not confusing it. You just don't seem to see it. `--force-overwrite` installs package that should be otherwise not installed. Take for example hostapd. Some packages need just plain hostapd so they depends on "virtual" package `hostapd` which is provided by all hostapd variants. But then there is package which needs features only in hostapd-full. Such package do not depends on `hostapd` but rather directly on `hostapd-full`. Installing `hostapd-full` with `--force-overwrite` makes both hostapd and hostapd-full installed. When in such case you do for example reinstall hostapd-mini it is still consistent from opkg view but anything depending on hostapd-full breaks. The worst case scenario is then with `opkg upgrade`. It upgrades all installed packages in order they are in status file. Question is in what order those packages are. Last upgraded package wins in this case. Is it hostapd-full or hostapd-mini? The solution are `Conflicts` and you are correctly arguing that they should be there. I will get to that.

> Upgrades are unsupported due to deficiencies in the package repo (lack of library ABI tracking. lack of proper maintenance of Supersedes/Replaces/Conflicts tags in package metadata, lack of central [...]

Turris is using same packages with updater-ng and not opkg and they work. I accept that ABI is problem there and we have some sort of solution for that but anything else is not. Packages are sensible enough that if you have sensible enough tool it works and I am not afraid to have it running from OpenWRT master updating daily. I am running that now for more than year and it works. The problem is not directly in packages them self.

Funny enough our experience is that `--force-overwrite` is the cause why updates break. Not for me, I don't use Luci, but for our users. So we have concrete proof that it is causing problems: https://forum.turris.cz/t/recommended-way-to-install-additional-packages/10338

> It is not random. The package getting installed will assume ownership of the conflicting files. If the partially overwritten package is removed later, the conflicting files are retained.

Problem is if it is removed later on. And if it should be removed. :wink: 

> Again. Neither do we ignore Depends: , not Conflicts: . Neither of these cases you outlined are affected by --force-overwrite.

You do not ignore depends but opkg is clearly ignoring `Conflicts:`. Proof on current OpenWRT 19.07 branch:
```
1:root@work-omnia:~$ opkg list-installed | grep wpad
wpad - 2018-12-02-c2c6c01b-6.0
root@work-omnia:~$ cat /usr/lib/opkg/info/wpad.control 
Package: wpad
Version: 2018-12-02-c2c6c01b-6.0
Depends: libc, libnl-tiny, hostapd-common, libubus
Conflicts: hostapd, hostapd-basic, hostapd-mini, hostapd-openssl, hostapd-wolfssl
Provides: hostapd, wpa-supplicant
Source: package/network/services/hostapd
SourceName: wpad
License: BSD-3-Clause
Section: net
Maintainer: Felix Fietkau <nbd@nbd.name>
Architecture: arm_cortex-a9_vfpv3
Installed-Size: 506956
Description:  This package contains a full featured IEEE 802.1x/WPA/EAP/RADIUS
 Authenticator and Supplicant
root@work-omnia:~$ opkg install hostapd
Installing hostapd (2018-12-02-c2c6c01b-6.0) to root...
Downloading https://repo.turris.cz/hbd/omnia/packages/base/hostapd_2018-12-02-c2c6c01b-6_arm_cortex-a9_vfpv3.ipk
Collected errors:
 * check_data_file_clashes: Package hostapd wants to install file /usr/sbin/hostapd
	But that file is already provided by package  * wpad
 * opkg_install_cmd: Cannot install package hostapd.
root@work-omnia:~$ opkg install --force-overwrite hostapd
Installing hostapd (2018-12-02-c2c6c01b-6.0) to root...
Downloading https://repo.turris.cz/hbd/omnia/packages/base/hostapd_2018-12-02-c2c6c01b-6_arm_cortex-a9_vfpv3.ipk
Configuring hostapd.
```

> There are no such solutions available in opkg. There is no atomic opkg replace ... operation available which would cleanly address this case.

I don't have cure for this. But I don't think that it is a valid reason to allow broken system.  
Our updater has this feature but it is too big to replace opkg in OWRT.

Cynerd, 2019-06-19:
Something I also forget to point out. Current OpenWRT has a lot of package where `Conflicts` are just not simply possible. Such package is for example busybox. It provides optionally so much tools that it is pretty much impossible to collect conflicts with rest of the system. These packages should be packaged differently so they are better manageable so in this case it is problem in packages but in current state we can't just say that we allow those packages to be broken.

jow-, 2019-06-19:
> Installing hostapd-full with --force-overwrite makes both hostapd and hostapd-full installed.

Yes but this is due to a bug in `opkg` (ignoring `Conflicts`) not due to the use of `--force-overwrite`. Or let me put it another way; if the file clash checking is the only thing preventing opkg from installing conflicting packages, despite a `Conflicts` tag present in control files, then we have a serious regression in opkg which needs fixing. Relying on file clash checks to handle conflicts is not the solution.

> You do not ignore depends but opkg is clearly ignoring Conflicts:. Proof on current OpenWRT 19.07 branch:

But this outside of the scope of this discussion, the `opkg` implementation should be fixed then to not ignore `Conflicts`.

jow-, 2019-06-19:
> Something I also forget to point out. Current OpenWRT has a lot of package where Conflicts are just not simply possible. Such package is for example busybox. It provides optionally so much tools [...]

This has been addressed with https://git.openwrt.org/?p=project/opkg-lede.git;a=commitdiff;h=21b7bd73ae88ba2ce9a5bfeedf67bc204d10fd08

Cynerd, 2019-06-19:
> Yes but this is due to a bug in opkg (ignoring Conflicts) not due to the use of --force-overwrite. Or let me put it another way; if the file clash checking is the only thing preventing opkg from [...]

I agree that there should be `Conflicts` where ever possible but it might not be always possible. It can be just because package maintainers do not know about such conflict. I have problem with hiding this kind of error.

The difference is that if there is explicit conflict then opkg can and should handle that with ease (well it does not currently but it is possible). If there is files conflict then this is fatal error in installation and bug in package. In both cases installation should not be permitted without explicit user agreement.

> This has been addressed with https://git.openwrt.org/?p=project/opkg-lede.git;a=commitdiff;h=21b7bd73ae88ba2ce9a5bfeedf67bc204d10fd08

I know about it but this addresses it only for busybox. :wink:

Anything that does not use links would be in such case installed multiple times. That somewhat goes against what OpenWRT stands for and also rest of the world is doing. Rest of the worlds is using `Provides` and `Conflicts` to replace alternative packages. I know that this is not possible with opkg at the moment but question is if rather than having ugly packages we should rather fix opkg.