openwrt/luci #2773: OpenWRT 18.06.2 DDNS NCM interface not shown

yurtesen, 2019-06-16:
I have setup 18.06.2 with ME909s-120 and using NCM.

In luci I see a `wwan_4` under interfaces. It says `Protocol: Virtual dynamic interface (DHCP client)` and it seems to have been created automatically.

When I go to DDNS settings at `Network [IPv4]` there is only `wwan` and no mention of `wwan_4`

I do need to set it up manually as `wwan_4` using ssh to get DDNS working properly. I wonder why `wwan_4` is not listed at ddns settings?

feckert, 2019-09-13:
I am not sure but I think it should work if you select the **wwan** interface in the dropdown

feckert, 2019-09-21:
@dibdot Could you please have a lool I am not familiar with ddns but I think this should work.

yurtesen, 2019-09-22:
@feckert no it does not work if I select wwan. Perhaps it has something to do with NCM and QMI and IP address not being propagated. 

See:
https://bugs.openwrt.org/index.php?do=details&task_id=1304

dibdot, 2019-09-22:
@feckert sorry, I can't help here as I don't use ddns scripts at all.

cmheia, 2019-10-27:
Try this if you could find `wwan0` list by run `ip a`:
ddns->Advanced Settings->IP address source: `Interface`
ddns->Advanced Settings->Interface: `wwan0`

Then ddns scripts should run `ip -o addr show dev wwan0 scope global` to detect local IP.
You could find `Detect local IP on 'interface'` in Log File Viewer.

aberhamm, 2024-12-28:
> Try this if you could find `wwan0` list by run `ip a`: ddns->Advanced Settings->IP address source: `Interface` ddns->Advanced Settings->Interface: `wwan0`
>
> Then ddns scripts should run `ip -o addr [...]

This worked for me. Thanks.