openwrt/luci #2774: `maxwait` and `delay` semantic confusion?

yurtesen, 2019-06-17:
I see in the `Advanced Settings` of interface for NCM says `Modem init timeout` and the description says `Maximum amount of seconds to wait for the modem to become ready`

But this text was taken from old ppp interface settings... can be seen in [this documentation page](https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols). While it is also unclear.

I am not even sure if it is used at all in openwrt code? as no references are found to it in any relevant script/code?:
[https://github.com/search?p=1&q=org%3Aopenwrt+maxwait&type=Code](https://github.com/search?p=1&q=org%3Aopenwrt+maxwait&type=Code)

Meanwhile `ncm` has a delay option which is used in [ncm.sh](https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/ncm.sh) also MBIM and QMI scripts use it.

This is the `delay` option of configuration as explained [here](https://openwrt.org/docs/guide-user/network/wan/wwan/ethernetoverusb_ncm) but there it says `Seconds to wait before trying to interact with the modem (some modems require up to 30 s.)`

When I was configuring NCM yesterday, I could not connect the dots... I did not understand that init timeout was waiting time before trying to turn on the interface.

Perhaps the maxwait should be removed as it does not seem to be doing anything and the text in luci should be updated to be more descriptive?

feckert, 2019-06-26:
> Perhaps the maxwait should be removed as it does not seem to be doing anything and the text in luci should be updated to be more descriptive?

That is correct maxwait will not be used https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/3g.sh so I will add a PR to remove this from Luci or change this to the delay function.

But I think we should add also `[ -n "$delay" ] && sleep "$delay"` to the proto 3g as this is already done in the protos NCM QMI MBIM

feckert, 2019-06-26:
See PR https://github.com/openwrt/openwrt/pull/2166 to add delay to the 3g proto.
See PR https://github.com/openwrt/luci/pull/2791 rename maxwait to delay

yurtesen, 2019-06-26:
Thank you for the quick response. I was also thinking that the correct text should be `Startup delay`  and description `Seconds to wait before trying to interact with the modem (some modems require up to 30s.)` as written exactly in wiki [https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols](https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols)

The word "timeout" gives the impression that how long system waits for modem to establish connection. In reality we are delaying before sending instructions to modem.

Furthermore, the current text says "maximum amount of seconds to wait" which is also misleading but goes in tandem with meaning of "timeout".  This setting is not  maximum number of seconds of waiting before timing out. In reality the delay is amount of seconds the script waits before sending instructions to modem.

I don't know if I could explain it clearly enough. I hope it was...

feckert, 2019-07-04:
The changes are now upstream.
Feel free to add a PR to change the translation.

Thanks for your feedback

closed 2019-07-04 by feckert