openwrt/luci #2781: Internal server error (unable to find section 'main')

Tomlincoln, 2019-06-21:
Hi Guys,

I think I have the same situation as in issue #1803 .
D-Link DWR-118 A2 fresh snapshot install tried hard reset etc etc.

/usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

ls -la /etc/config/luci
[code]
-rw-------    1 root     root           793 Jun 21 14:38 /etc/config/luci
[/code]

grep "main" /etc/config/luci
[code]
config core 'main'
[/code]

opkg list-installed luci*
[code]
luci - git-19.170.41626-1154c41-1
luci-app-firewall - git-19.170.41626-1154c41-1
luci-app-opkg - git-19.170.41626-1154c41-1
luci-base - f138fc93-1
luci-lib-ip - git-19.170.41626-1154c41-1
luci-lib-jsonc - git-19.170.41626-1154c41-1
luci-lib-nixio - git-19.170.41626-1154c41-1
luci-mod-admin-full - git-19.170.41626-1154c41-1
luci-mod-network - git-19.170.41626-1154c41-1
luci-mod-status - git-19.170.41626-1154c41-1
luci-mod-system - git-19.170.41626-1154c41-1
luci-proto-ipv6 - git-19.170.41626-1154c41-1
luci-proto-ppp - git-19.170.41626-1154c41-1
luci-theme-bootstrap - git-19.170.32094-4d6d8bc-1
[/code]

ls -l /usr/share/rpcd/acl.d/
[code]
-rw-r--r--    1 root     root           159 Jun 20 13:48 luci-base.json
-rw-r--r--    1 root     root           179 Jun  7 16:22 unauthenticated.json
[/code]

ls -l /usr/share/acl.d/
[code]
-rw-r--r--    1 root     root            50 Jun 21 10:03 dnsmasq_acl.json
-rw-r--r--    1 root     root            91 Jun 20 13:48 luci-base.json
[/code]

luci config file

config core 'main'
        option lang 'auto'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
        option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
        option sessionpath '/tmp/luci-sessions'
        option sessiontime '3600'

config internal 'ccache'
        option enable '1'

config internal 'themes'

config internal 'apply'
        option rollback '30'
        option holdoff '4'
        option timeout '5'
        option display '1.5'

config internal 'diag'
        option dns 'openwrt.org'
        option ping 'openwrt.org'
        option route 'openwrt.org'

jow-, 2019-06-21:
Ensure that `rpcd` is running. If it is a custom build, run `make dirclean` before compiling a new firmware.

closed 2019-06-21 by jow-

antoan-boss, 2019-06-21:
Same issue here with latest snapshot for ZBT-WE1326. I checked that rpcd is not running and it can't start...
rpcd
Error loading shared library libjson-c.so.2: No such file or directory (needed by /sbin/rpcd)

I did not find such file, only libjson-c.so.4
There is libjson-c4 - 0.13.1-1 installed probably newer than before and without libjson-c.so.2. :/

Tomlincoln, 2019-06-21:
rpcd (ubusd) is installed and running:

root@OpenWrt:~# opkg list ubus*
```
ubus - 2018-10-06-221ce7e7-1 - OpenWrt RPC client utility
ubusd - 2018-10-06-221ce7e7-1 - OpenWrt RPC daemon
```

root@OpenWrt:~# ps | grep ubus
  ```
504 root      1212 S    /sbin/ubusd
 3294 root      1200 S    grep ubus
```

This is not a custom build, it's the latest snapshot, and tried opkg install luci.

jow-, 2019-06-21:
Nothing we can fix on the LuCI side. Need to wait a few days for snapshot builds to stabilize. Alternatively consider building a custom firmware image.

antoan-boss, 2019-06-21:
Installed and running..

root@OpenWrt:~# opkg list ubus*
```
ubus - 2018-10-06-221ce7e7-1 - OpenWrt RPC client utility
ubusd - 2018-10-06-221ce7e7-1 - OpenWrt RPC daemon
```
root@OpenWrt:~# ps | grep ubus
```
  554 root      1216 S    /sbin/ubusd
 2772 root      1204 S    grep ubus
```

What about this? It's ok?
root@OpenWrt:/# rpcd
`Error loading shared library libjson-c.so.2: No such file or directory (needed by /sbin/rpcd)`

Yea looks like not a LuCI issue...

Tomlincoln, 2019-06-21:
Checked, having the exact  same issue with libjson!

Tomlincoln, 2019-06-22:
Anybody who reads this, the **solution is the following**:
Link the new library to on the old name in /usr/lib:

```
root@OpenWrt:~# cd /usr/lib
root@OpenWrt:/usr/lib# ln -s libjson-c.so.4.0.0 libjson-c.so.2
```

odon-noda, 2019-09-07:
**Installing rpcd has helped in my case.**

root@OpenWrt:/usr/lib# /etc/init.d/rpcd restart
Command failed: Not found
root@OpenWrt:/usr/lib# opkg install rpcd
Upgrading rpcd on root from 2019-06-05-89bfaa42-3 to 2019-09-01-821045f6-3...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/rpcd_2019-09-01-821045f6-3_mips_24kc.ipk
Configuring rpcd.
root@OpenWrt:/usr/lib# ls -la libjs*
lrwxrwxrwx    1 root     root            18 Sep  5 12:33 libjson-c.so.4 -> libjson-c.so.4.0.0
-rwxr-xr-x    1 root     root         41752 Sep  5 12:33 libjson-c.so.4.0.0
root@OpenWrt:/usr/lib# /etc/init.d/rpcd restart
root@OpenWrt:/usr/lib# /etc/init.d/uhttpd restart
root@OpenWrt:/usr/lib# /etc/init.d/uhttpd enable

The possible cause was somewhere outside the router.
I installed two ath79 in the same way. First yesterday, the second today. First worked.
Luci failed on the second one.
Installing rpcd on the second one helped.

seandex, 2019-10-15:
@odon-noda confirmed. it is rpcd problem.
cd /tmp
wget http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/rpcd_2019-09-01-821045f6-3_mips_24kc.ipk
opkg install rpcd_2019-09-01-821045f6-3_mips_24kc.ipk --force-downgrade
worked.