openwrt/luci #2787: Wireless doesn't detect uncommited uci settings

hanipouspilot, 2019-06-25:
Recently I noticed that the wireless tab doesn't show normally disabled interface when I bring it up by UCI. I didn't notice when this happened. The tab doesn't show "uncommited settings" as usual.

I found it on 19.07 branch. It used to work on master before.

jow-, 2019-06-25:
There are no changes in openwrt-19.07 compared to master in the LuCI repo, which hints at a bug in OpenWrt instead.

hanipouspilot, 2019-06-25:
I noticed that when I enable an interface using "Enable" button in Luci, the change automatically is committed. Maybe didn't notice it in master. It didn't work this way before.

UCI works normally the setting is not applied to /etc/config/wireless, but interface works. So it looks like Luci reads the config files instead of uci temp.

jow-, 2019-06-26:
Correct, LuCI master and 19.07 will use separate per-session uci delta directories and ignore the system wide uci temp.

closed 2019-06-26 by jow-