openwrt/luci #2797: status/processes page missing all "niced" processes

hnyman, 2019-06-27:
@jow- 

I noticed that the status/processes page is missing all "niced" processes (which currently includes e.g. collectd and nlbwmon) as well as ntpd.

I think that the reason is lua pattern matching, as the niced processes are still in the busybox top output, but not in the final LuCI page. That does not look purposeful to me, as currently some "normal" monitor-style processes have been niced, but they should still be visible here as they are normal use processes.

I think that it is due to the pattern at:

https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/sys.lua#L446
```
^ *(%d+) +(%d+) +(%S.-%S) +([RSDZTW][W ][<N ]) +(%d+) +(%d+%%) +(%d+%%) +(.+)"
```

The output from top is:
```
root@router1:~# /bin/busybox top -bn1
Mem: 137300K used, 340088K free, 1696K shrd, 6072K buff, 20180K cached
CPU:   0% usr   4% sys   0% nic  95% idle   0% io   0% irq   0% sirq
Load average: 0.05 0.01 0.00 3/76 12553
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 4091     1 root     S     4184   1%   1% /usr/sbin/hostapd -s -P ...
 4565     1 root     SN    1116   0%   1% /usr/sbin/nlbwmon -o /var/lib/nlbwmon ...
12553 12380 root     R     1084   0%   1% {top} /bin/busybox top -bn1
    8     2 root     IW       0   0%   1% [rcu_sched]
 4808     1 root     SN    4616   1%   0% /usr/sbin/collectd -C /tmp/collectd.conf -f
 4160     1 root     S     4184   1%   0% /usr/sbin/hostapd -s -P ...
...
 2784     1 dnsmasq  S     1324   0%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg...
 1689     1 root     S     1240   0%   0% /usr/sbin/odhcpd
12380 12379 root     S     1084   0%   0% -ash
 9314     1 root     S<    1076   0%   0% /usr/sbin/ntpd -n -N -S ...
...

```

The pattern used for matching has been the same since 6 years ago in Jan 2013 by 
https://github.com/openwrt/luci/commit/a52a6a4f22861bbb1b7676b91b1f873e09bdbdfd#diff-cb4cb2401e6e43cf74a54523b8da5f02

I am no lua expert but the current pattern "[RSDZTW][W ][<N ]" looks like it rejects processes where there is no W or space before N or <.

I wonder if the middle part of the pattern should include also "<N", so that it would be something like "[RSDZTW][<NW ][<N ]".

Making that change brings collectd, nlbwmon and ntpd visible on the LuCI page.

Does that logic and change sound correct?

jow-, 2019-06-28:
Sounds about right. Can you make a PR for it?

dibdot, 2019-06-28:
while at it we could maybe remove the top process itself from output!? 
Change line 438 to `	local ps = luci.util.execi("/bin/busybox top -bn1 | grep -v 'top -bn1'")`

jow-, 2019-06-28:
Please filter in Lua code, otherwise you'll have another `grep` process to filter out ;)

dibdot, 2019-06-28:
@hnyman it might be better to add some logic in line 450, e.g:

`		if idx and not cmd:match("top %-bn1") then`

hnyman, 2019-06-28:
Thanks for feedback and suggestions.

Fixed with 1c80eb1962

closed 2019-06-28 by hnyman