openwrt/luci #2798: [feature suggestion] add support for DSA switch configuration

n8v8R, 2019-06-27:
* trunk
___
 With the transition from `swconfig` to DSA it is observed that (logically in view of the dependency) the switch section in LuCI is a no show. Hence, this feature suggestion.

Unless I am mistaken, UCI is currently limited about what DSA settings can be parsed from a conf file in comparison to `swconfig`, e.g. it seems to be lacking to support for setting DSA 

* untag state
* off state
* PVID assignment

Discovery from https://lore.kernel.org/patchwork/patch/575746/ would suggest that such is achieved with `bridge vlan` but maybe there are other ways.

jow-, 2019-06-27:
This needs to be supported in OpenWrt's /etc/config/network first. I have no idea how to approach that form the LuCI side.

n8v8R, 2019-06-27:
Does it require opening a task in Flyspray for `netifd`?

jow-, 2019-06-27:
Maybe. First you need to figure out which parts are actually missing and which are supported already.

n8v8R, 2019-06-27:
If you could hint of how or where to look for. I have hit a roadblock in this regard by perusing the available online documentation.

jow-, 2019-06-27:
Me too :) I have no idea how this DSA is supposed to work. It was always being told that we just have to bridge interfaces together now. I would need to learn how to setup "off state", "untag state" and PVIDs on a normal linux system without uci or OpenWrt specialities, then I can look into how to translate that to uci.

However, I'm not sure whether there'll ever be a swconfig-like configuration page for DSA in the near future. It is totally unclear how to approach that, how to figure out if a netdev belongs to a DSA switch etc.

Right now the topology information isn't available. This all needs to be sorted out. I have no access to DSA capable hardware, so it is up to others to assemble the required info.

n8v8R, 2019-06-27:
`bridge vlan` appears to working, which is not OpenWRT specific.

Other than the afore cited discovery I have drawn mostly blanks.

n8v8R, 2019-06-27:
perhaps `ubus` can query that info. say at least `ip -d l | egrep switch` produces

> bridge_slave state forwarding priority 32 cost 4 hairpin off guard off root_block off fastleave off learning on flood on port_i0x8001 port_no 0x1 designated_port 32769 designated_cost 0 designated_bridge 7fff.D8:58:D7:00:79:7A designated_roo7fff.D8:58:D7:00:79:7A hold_timer    0.00 message_age_timer    0.00 forward_delay_timer    0.00 topology_change_ack 0 config_pendin0 proxy_arp off proxy_arp_wifi off mcast_router 1 mcast_fast_leave off mcast_flood on neigh_suppress off vlan_tunnel off addrgenmodstable_secret numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 portname p0 switchid 00000000
> bridge_slave state forwarding priority 32 cost 4 hairpin off guard off root_block off fastleave off learning on flood on port_i0x8002 port_no 0x2 designated_port 32770 designated_cost 0 designated_bridge 7fff.D8:58:D7:00:79:7A designated_roo7fff.D8:58:D7:00:79:7A hold_timer    0.00 message_age_timer    0.00 forward_delay_timer    0.00 topology_change_ack 0 config_pendin0 proxy_arp off proxy_arp_wifi off mcast_router 1 mcast_fast_leave off mcast_flood on neigh_suppress off vlan_tunnel off addrgenmodstable_secret numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 portname p1 switchid 00000000
> bridge_slave state disabled priority 32 cost 100 hairpin off guard off root_block off fastleave off learning on flood on port_i0x8003 port_no 0x3 designated_port 32771 designated_cost 0 designated_bridge 7fff.D8:58:D7:00:79:7A designated_roo7fff.D8:58:D7:00:79:7A hold_timer    0.00 message_age_timer    0.00 forward_delay_timer    0.00 topology_change_ack 0 config_pendin0 proxy_arp off proxy_arp_wifi off mcast_router 1 mcast_fast_leave off mcast_flood on neigh_suppress off vlan_tunnel off addrgenmodstable_secret numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 portname p2 switchid 00000000
> bridge_slave state disabled priority 32 cost 100 hairpin off guard off root_block off fastleave off learning on flood on port_i0x8001 port_no 0x1 designated_port 32769 designated_cost 0 designated_bridge 7fff.06:F3:2D:B4:66:17 designated_roo7fff.06:F3:2D:B4:66:17 hold_timer    0.00 message_age_timer    0.00 forward_delay_timer    0.00 topology_change_ack 0 config_pendin0 proxy_arp off proxy_arp_wifi off mcast_router 1 mcast_fast_leave off mcast_flood on neigh_suppress off vlan_tunnel off addrgenmodstable_secret numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 portname p3 switchid 00000000
> bridge_slave state disabled priority 32 cost 100 hairpin off guard off root_block off fastleave off learning on flood on port_i0x8001 port_no 0x1 designated_port 32769 designated_cost 0 designated_bridge 7fff.D8:58:D7:00:79:7A designated_roo7fff.D8:58:D7:00:79:7A hold_timer    0.00 message_age_timer    0.00 forward_delay_timer    0.00 topology_change_ack 0 config_pendin0 proxy_arp off proxy_arp_wifi off mcast_router 1 mcast_fast_leave off mcast_flood on neigh_suppress off vlan_tunnel off addrgenmodstable_secret numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 portname p4 switchid 00000000

n8v8R, 2019-06-27:
I would not mind in whatever way feasible to assist development, considering owning a DSA device, say test scripts

n8v8R, 2019-06-28:
As far as I can discern UCI (in conjunction with netifd?) upon

`uci set network.guest.ifname='lan4.1'`

invokes

`ip l a l lan4 n lan4.1 type vlan id 1`

[*] br-guest enslaves lan4.1 in this use case
___

`ip l a l [iface.N] n [vlan] type vlan id X` appears to be limited to just enable VLAN with tag state but offers no advanced VLAN management capabilities. Suppose that is where `bridge vlan` comes in.
___
`ubus call network.interface dump` does not appear to be a suitable tool to query the DSA topology.

perhaps `ls -l /sys/class/net | egrep mdio_bus`

> lan0 -> ../../devices/platform/soc/soc:internal-regs/f1072004.mdio/mdio_bus/f1072004.mdio-mii/f1072004.mdio-mii:10/net/lan0
> lan1 -> ../../devices/platform/soc/soc:internal-regs/f1072004.mdio/mdio_bus/f1072004.mdio-mii/f1072004.mdio-mii:10/net/lan1
> lan2 -> ../../devices/platform/soc/soc:internal-regs/f1072004.mdio/mdio_bus/f1072004.mdio-mii/f1072004.mdio-mii:10/net/lan2
> lan3 -> ../../devices/platform/soc/soc:internal-regs/f1072004.mdio/mdio_bus/f1072004.mdio-mii/f1072004.mdio-mii:10/net/lan3
> lan4 -> ../../devices/platform/soc/soc:internal-regs/f1072004.mdio/mdio_bus/f1072004.mdio-mii/f1072004.mdio-mii:10/net/lan4

n8v8R, 2019-06-28:
reading about man `bridge vlan` https://manpages.debian.org/experimental/iproute2/bridge.8.en.html#bridge_vlan_-_VLAN_filter_list

it likely requires:

* kernel config `CONFIG_BRIDGE_VLAN_FILTERING=y`, plus
* sysctl `/sys/class/net/[bridge-name]/bridge/vlan_filtering = 1`

n8v8R, 2019-06-29:
Done a bit research in the public domain for 

* `linux manage 802.1q`
* `iproute2 ip link 802.1q`
* `iproute2 cheat sheet`

and reached the conclusion  that

`ip l a l ${interface name} n ${virtual interface}.{name} type vlan id {VID}`

is inhibited to just enabling (and in reverse disabling) 802.1q with tag state only but lacks advanced management of 802.1q. 

see https://www.linux.org/docs/man8/ip-link.html

> vlan - 802.1q tagged virtual LAN interface

> VLAN Type Support
> * reorder_hdr { on | off } - specifies whether ethernet headers are reordered or not.
> * gvrp { on | off } - specifies whether this VLAN should be registered using GARP VLAN Registration Protocol.
> * mvrp { on | off } - specifies whether this VLAN should be registered using Multiple VLAN Registration Protocol.
> * loose_binding { on | off } - specifies whether the VLAN device state is bound to the physical
> device state.
> * ingress-qos-map QOS-MAP - defines a mapping between priority code points on incoming frames.
> The format is FROM:TO with multiple mappings separated by spaces.
> * egress-qos-map QOS-MAP - the same as ingress-qos-map but for outgoing frames.

Only `bridge vlan` appears to have the capability but I would be glad if someone would share a different methodology.

n8v8R, 2019-06-30:
a use case basic example for untag 802.1q frames, br-lan enslaving lan0,lan1,lan2

```
bridge v a dev br-lan self vid 2 untagged pvid
bridge v a dev lan0 vid 2 untagged pvid
bridge v a dev lan1 vid 2 untagged pvid
bridge v a dev lan2 vid 2 untagged pvid
sysctl -w /sys/class/net/br-lan/bridge/vlan_filtering=1
```

`sysctl -w` sets the filter in motion and thus to be last or else access to the node could be cut off.

Diman779, 2019-07-15:
https://github.com/login?return_to=%2Faccount%2Funverified-email

Diman779, 2019-07-15:
activ.kz/ru

CHKDSK88, 2019-07-24:
I'm working on this feature for  base OpenWRT. There is problem with some DSA switches, which don't have dsa tagging support and vlan filtering is only way to support it.

My solution will base on ip-bridge utility. I will put it for RFC when it's done.

n8v8R, 2019-07-24:
@CHKDSK88 perhaps worth for the base development to take into consideration - observing during boot on a node with 19.07 target mvebu:

> mv88e6085 f1072004.mdio-mii:10 lan0: configuring for phy/gmii link mode
> mv88e6085 f1072004.mdio-mii:10: p0: hw **VLAN 1 already used** by br-guest
> mv88e6085 f1072004.mdio-mii:10 lan1: configuring for phy/gmii link mode
> mv88e6085 f1072004.mdio-mii:10: p1: hw **VLAN 1 already used** by br-guest
> mv88e6085 f1072004.mdio-mii:10 lan2: configuring for phy/gmii link mode
> mv88e6085 f1072004.mdio-mii:10: p2: hw **VLAN 1 already used** by br-guest
> mv88e6085 f1072004.mdio-mii:10 lan3: configuring for phy/gmii link mode
> mv88e6085 f1072004.mdio-mii:10: p3: hw **VLAN 1 already used** by br-guest

which would appear to be generated by https://elixir.bootlin.com/linux/v4.14.132/source/drivers/net/dsa/mv88e6xxx/chip.c#L1157

n8v8R, 2019-08-03:
With regard to the aforementioned 

> VLAN 1 already used by

I came across  https://patchwork.ozlabs.org/patch/572082/ citing

> Register 7 for the ports is the default vlan tag register

and

https://android.googlesource.com/kernel/msm/+/android-msm-hammerhead-3.4-kk-r1/drivers/net/dsa/mv88e6131.c#248

```
 * Default VLAN ID and priority: don't set a default VLAN
	 * ID, and set the default packet priority to zero.
	 */
	REG_WRITE(addr, 0x07, 0x0000);
```

n8v8R, 2019-11-14:
Adding upstream documentation https://www.kernel.org/doc/html/latest/networking/dsa/configuration.html

n8v8R, 2019-12-28:
Brief summary about the current state.
1. UCI | LuCI does support parsing:

* `ip l a l ${interface name} n ${interface name}.{virtual extension} type vlan id [numeric value]`

which always creates a virtual interface that is 802.1q tagged. It can be leveraged with the node's upstream and downstream hardware interfaces. Removal of such virtual interface also works.
Managing tag/untag state with related VID/PVID of existing downstream interfaces does not work with the `ip` command by design.

2. UCI | LuCI does not support parsing:

* `ip l s dev ${bridge name} ty bridge vlan_filtering [ numeric value 0 | 1]`
and/or
 * `echo [ numeric value 0 | 1]  > /sys/class/net/${bridge name}/bridge/vlan_filtering`

either of which is leveraged to toggle VLAN filtering (for bridge isolation) on or off.

*  `bridge v a dev ${interface name} vid [numeric value] [options]`
*  `bridge v d dev ${interface name} vid [numeric value] [options]`

used for the management of 802.1q tag/untag state and related VID/PVID of hardware downstream interfaces (lan and bridge) and does not create any virtual interface.

n8v8R, 2019-12-29:
Since `ip` and `bridge` both feature batch operation with `-b` such might be useful to leverage to store:

* 802.1q tag/untag state and related VID/PVID of downstream interfaces (lan and bridge)

example to wipe the default VID 1 that is being assigned to all bridge and lan devices

`bridge -b /etc/config/vlan` with the file content

```
v d dev lan0 vid 1
v d dev lan1 vid 1
v d dev lan2 vid 1
v d dev lan3 vid 1
v d dev lan4 vid 1
v d dev br-lan vid 1 self
v d dev br-guest vid 1 self
```

* bridge isolation toggle

`ip -b /etc/config/br_isolation` with file content

```
l s dev br-lan ty bridge vlan_filtering 1
l s dev br-guest ty bridge vlan_filtering 0
```

n8v8R, 2020-01-02:
On a device with _dual/multiple CPUs_ toggling VLAN awareness on with `ip l s dev ${bridge name} ty bridge vlan_filtering 1` could lead to loss of access/connectivity to the router unless a proper Multi-CPU-DSA patch is applied.

n8v8R, 2020-02-01:
On  a side node - it appears that 802.11ac interfaces are also exposed to 802.1q tag management via `bridge v`, e.g. on an instance with:

* Master with kernel 4.19.93 on mvebu with DSA
* ip-bridge 5.4.0
* ath10k_pci 4.19.93

they exhibited as

> WLan_5G 1 PVID Egress Untagged
> WLan_5G_Guest 1 PVID Egress Untagged

and respond to

```
bridge v d dev WLan_5G_Guest vid 1
bridge v a dev WLan_5G_Guest vid 10
```
Perhaps worth considering the include in LuCI's 802.1q tag management code

n8v8R, 2020-07-09:
some progress https://patchwork.ozlabs.org/project/openwrt/patch/20200707201035.751527-2-jo@mein.io/ to lay the groundwork for LuCI :+1:

Shine-, 2020-07-23:
Nice to see. Also note the mailing list discussion that's going on about the proposal: http://lists.openwrt.org/pipermail/openwrt-devel/2020-July/029955.html

Looks like it's going into a direction that hopefully moves away from the painful syntax that UCI uses for swconfig! Fingers crossed!

youxiaojie, 2021-01-29:
support
does any solutions for dual cpu ports?

youxiaojie, 2021-01-29:
what's the difference with internel portbased vlan and 802.11Q? THE default dsa drivers' action is forward all ports to and from cpu port.

jow-, 2021-03-19:
DSA config support merged with https://github.com/openwrt/luci/commit/4cde2bda732b133d0ea46cecd94f57c3b1eeab05

closed 2021-03-19 by jow-