openwrt/luci #2799: /usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'

vdemtcev, 2019-06-27:
We use  http://downloads.openwrt.org/snapshots/targets/ar71xx/mikrotik/openwrt-ar71xx-mikrotik-rb-nor-flash-16M-squashfs-sysupgrade.bin on MikroTik RouterBOARD wAP R-2nD
After install luci and login to web interface we have an error:
/usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

and error log on device^
Thu Jun 27 17:55:24 2019 kern.info kernel: [  214.558530] do_page_fault(): sending SIGSEGV to rpcd for invalid read access from 00000020
Thu Jun 27 17:55:24 2019 kern.info kernel: [  214.567151] epc = 77e4f463 in libjson-c.so.4.0.0[77e4c000+19000]
Thu Jun 27 17:55:24 2019 kern.info kernel: [  214.573397] ra  = 77e69079 in libblobmsg_json.so[77e68000+11000]
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: luci: failed login on / for root from 10.50.50.254
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: /usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: stack traceback:
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: 	[C]: in function 'assert'
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: 	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
Thu Jun 27 17:55:54 2019 daemon.err uhttpd[2502]: 	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

Thu Jun 27 17:57:51 2019 daemon.err uhttpd[2502]: /usr/lib/lua/luci/dispatcher.lua:247: /etc/config/luci seems to be corrupt, unable to find section 'main'
Thu Jun 27 17:57:51 2019 daemon.err uhttpd[2502]: stack traceback:
Thu Jun 27 17:57:51 2019 daemon.err uhttpd[2502]: 	[C]: in function 'assert'
Thu Jun 27 17:57:51 2019 daemon.err uhttpd[2502]: 	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
Thu Jun 27 17:57:51 2019 daemon.err uhttpd[2502]: 	/usr/lib/lua/luci/dispatcher.lua:144: in function </usr/lib/lua/luci/dispatcher.lua:143>

it is NOT a Luci problem! Bug in 'rpcd' (but you read it like Bug in 'libjson-c.so.4.0.0' and 'libblobmsg_json.so')

Dancing with a tambourine like 'ln -s libjson-c.so.4.0.0 libjson-c.so.2' do nothing, cuz:
root@OpenWrt:/usr/lib# ls -la | grep libjs
lrwxrwxrwx    1 root     root            18 Jun 17 04:53 libjson-c.so.2 -> libjson-c.so.2.0.2
-rwxr-xr-x    1 root     root         29208 Jun 17 04:53 libjson-c.so.2.0.2
lrwxrwxrwx    1 root     root            18 Jun 17 04:55 libjson-c.so.4 -> libjson-c.so.4.0.0
-rwxr-xr-x    1 root     root         41752 Jun 26 05:41 libjson-c.so.4.0.0

And what we do? Maybe someone know how fix it?

vdemtcev, 2019-06-27:
same issue
https://github.com/openwrt/luci/issues/2781
https://github.com/openwrt/luci/issues/1803

vdemtcev, 2019-06-27:
cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r10243-4d11c4c378'
DISTRIB_TARGET='ar71xx/mikrotik'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r10243-4d11c4c378'
DISTRIB_TAINTS=''

jow-, 2019-06-27:
Nothing we can solve in the LuCI repo. Note that ar71xx is discontinued in OpenWrt master. Either switch to ath79 if possible or stack to the 18.06.x or upcoming OpenWrt 19.07 release series when released.

closed 2019-06-27 by jow-

vdemtcev, 2019-06-27:
Ok, i create bug report on OpenWRT https://bugs.openwrt.org/index.php?do=details&task_id=2343
But it is a big trouble, not only Mikrotik. And for many it will look like a big in Luci. Please, Immerse yourself deeper into this problem because rpcd is a critical element for Luci.

locojohn, 2020-01-23:
I am having exact same issue with the latest OpenWrt snapshot on ipq806x platform (NETGEAR R7800):

NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r12091-a66a1eb9c5"
OPENWRT_BOARD="ipq806x/generic"
OPENWRT_ARCH="arm_cortex-a15_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r12091-a66a1eb9c5"

mh4nx7net, 2020-06-17:
[solved here](https://github.com/openwrt/luci/issues/2885 )
hope help

xztxy, 2021-01-05:
ssh  use  "/etc/init.d/rpcd restart"

Raphael2time, 2022-04-26:
> ssh use "/etc/init.d/rpcd restart"

this canrestore

Raphael2time, 2022-04-26:
> ssh use "/etc/init.d/rpcd restart"

this can restore

ChenYuJie1987, 2022-07-04:
<html>
<body>

rm -r /tmp/luci-indexcache
--

</body>
</html>