openwrt/luci #2801: [feature suggestion] be clear about  802.1q management settings in switch config

n8v8R, 2019-06-28:
* trunk
___
The language semantics in the switch config page refers solely to VLAN, which to my understanding presents basically just L2 connectivity whilst port states `tag` | `untag` | `off` and VID/PVID assignment pertain instead to the management of the 802.1q protocol.

jow-, 2019-06-28:
Please translate this into actionable items.

n8v8R, 2019-06-28:
Assuming that `Enable VLAN functionality` just does that, i.e. enabling L2 connectivity, it could be retained as is. If it however also enables `802.1q` perhaps amened it to `Enable VLAN functionality with 802.1q protocol`

Else for `VLANs on` I would humbly suggest substituting it with `802.1q protocol settings` or `802.1q protocol management`

Unless it is considered too technical of hair splitting and confusing the user since public domains seem often to refer to `802.1q` as  `VLAN tagging` of sorts and more than often lack the distinction.