openwrt/luci #2807: Unable to launch the requested CGI program: /www/cgi-bin/luci: No such file or directory

intdev32, 2019-06-30:
Web browser, 192.168.1.1 enter then,

> Unable to launch the requested CGI program: 
> 
> /www/cgi-bin/luci: No such file or directory
> 

Version Information:
luci - git-19.180.62908-a04e22e-1
 OpenWrt SNAPSHOT, r10367-9e780ed5f7

--
Just 1 hour earlier, I remember luci version is 19.178.xxxxx. But now, luci version updated? opkg shows luci version is 19.180.

Just 1 hour earlier, Luci worked well. (19.178.xxxxx)

My 2 OpenWrt devices, same results. (I retried Hard-Reset...)

intdev32, 2019-06-30:
Router Factory reset.

I changed opkg repo. to russia. http://mirrors.linux.ro/lede/downloads/snapshots/
(Yesterday mirror state. Still not updated to today.)

opkg update && opkg install luci

and checked 192.168.1.1.

Nicely working...

There are somethings wrong definitely!

intdev32, 2019-06-30:
I found solution.

opkg install luci.

After,

opkg install lua

Now works.

*** Luci installs lua5.3. Not lua.

-----------
root@OpenWrt:~# opkg update && opkg install luci
Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/kmods/4.14.131-1-d2c6f311e7cd7563c0e0665b6cc54366/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/kmods/4.14.131-1-d2c6f311e7cd7563c0e0665b6cc54366/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/telephony/Packages.sig
Signature check passed.
Installing luci (git-19.180.62908-a04e22e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci_git-19.180.62908-a04e22e-1_all.ipk
Installing uhttpd (2019-06-16-91fcac34-3) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/uhttpd_2019-06-16-91fcac34-3_arm_xscale.ipk
Installing lua5.3 (5.3.5-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/lua5.3_5.3.5-1_arm_xscale.ipk
Installing liblua5.1.5 (5.1.5-4) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/liblua5.1.5_5.1.5-4_arm_xscale.ipk
Installing luci-lib-nixio (git-19.180.62908-a04e22e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-nixio_git-19.180.62908-a04e22e-1_arm_xscale.ipk
Installing luci-lib-ip (git-19.180.62908-a04e22e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-ip_git-19.180.62908-a04e22e-1_arm_xscale.ipk
Installing rpcd (2019-06-05-89bfaa42-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/rpcd_2019-06-05-89bfaa42-2_arm_xscale.ipk
Installing libubus-lua (2018-10-06-221ce7e7-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/libubus-lua_2018-10-06-221ce7e7-1_arm_xscale.ipk
Installing luci-lib-jsonc (git-19.180.62908-a04e22e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-jsonc_git-19.180.62908-a04e22e-1_arm_xscale.ipk
Installing liblucihttp0 (2019-06-05-91c01c3c-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/liblucihttp0_2019-06-05-91c01c3c-1_arm_xscale.ipk
Installing liblucihttp-lua (2019-06-05-91c01c3c-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/liblucihttp-lua_2019-06-05-91c01c3c-1_arm_xscale.ipk
Installing luci-base (f138fc93-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-base_f138fc93-1_arm_xscale.ipk
Installing libiwinfo-lua (2019-06-12-1372f47e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/libiwinfo-lua_2019-06-12-1372f47e-1_arm_xscale.ipk
Installing luci-mod-status (git-19.180.62908-a04e22e-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-mod-status_git-19.180.62908-a04e22e-1_arm_xscale.ipk
[... 21 lines cut ...]
Configuring liblua5.1.5.
Configuring luci-lib-ip.
Configuring libubus-lua.
Configuring libiwinfo-lua.
Configuring lua5.3.
Configuring luci-lib-nixio.
Configuring luci-lib-jsonc.
Configuring liblucihttp-lua.
Configuring luci-base.
Configuring luci-mod-system.
Configuring luci-theme-bootstrap.
Configuring luci-mod-status.
Configuring luci-app-firewall.
Configuring luci-proto-ppp.
Configuring luci-mod-network.
Configuring luci-mod-admin-full.
Configuring luci-proto-ipv6.
Configuring uhttpd.
Configuring rpcd-mod-rrdns.
Configuring luci.
root@OpenWrt:~# opkg install lua
Installing lua (5.1.5-4) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/lua_5.1.5-4_arm_xscale.ipk
Configuring lua.
root@OpenWrt:~#

hnyman, 2019-06-30:
@jow-  @rmilecki 

 intdev32 commented 15 minutes ago :
> I found solution:
>  *** Luci installs lua5.3. Not lua.

> opkg install luci.
> Configuring lua5.3.
> ...does not work

> After,
> opkg install lua
> Installing lua (5.1.5-4) to root...
> Now works.

Based on that finding, apparently the introduction of lua5.3 with https://github.com/openwrt/openwrt/commit/6b161bb8d528663b78c2314d9f1e1c9217dacba2 by @rmilecki into the main OpenWrt repository breaks installing LuCI to a live system.

Likely because lua5.3 includes a ` PROVIDES:=lua ` statement, which may cause opkg to install that instead of the proper normal lua.

Maybe the normal lua should be declared the as DEFAULT_VARIANT ?

rmilecki, 2019-06-30:
On Sun, 30 Jun 2019 at 09:34, Hannu Nyman <notifications@github.com> wrote:
> I found solution:
> *** Luci installs lua5.3. Not lua.

Please try 17ae3eb9ff9d ("lua5.3: drop unwanted & unneeded PROVIDES").

rfariavitor, 2019-06-30:
> I found solution.
> 
> opkg install luci.
> 
> After,
> 
> opkg install lua
> 
> Now works.
> 
> *** Luci installs lua5.3. Not lua.
> 
> `root@OpenWrt:~# opkg update && opkg install luci
> Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_core
> Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/kmods/4.14.131-1-d2c6f311e7cd7563c0e0665b6cc54366/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_kmods
> Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/kmods/4.14.131-1-d2c6f311e7cd7563c0e0665b6cc54366/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_base
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_luci
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/packages/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_packages
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/packages/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/routing/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_routing
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/routing/Packages.sig
> Signature check passed.
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/telephony/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_telephony
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/telephony/Packages.sig
> Signature check passed.
> Installing luci (git-19.180.62908-a04e22e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci_git-19.180.62908-a04e22e-1_all.ipk
> Installing uhttpd (2019-06-16-91fcac34-3) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/uhttpd_2019-06-16-91fcac34-3_arm_xscale.ipk
> Installing lua5.3 (5.3.5-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/lua5.3_5.3.5-1_arm_xscale.ipk
> Installing liblua5.1.5 (5.1.5-4) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/liblua5.1.5_5.1.5-4_arm_xscale.ipk
> Installing luci-lib-nixio (git-19.180.62908-a04e22e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-nixio_git-19.180.62908-a04e22e-1_arm_xscale.ipk
> Installing luci-lib-ip (git-19.180.62908-a04e22e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-ip_git-19.180.62908-a04e22e-1_arm_xscale.ipk
> Installing rpcd (2019-06-05-89bfaa42-2) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/rpcd_2019-06-05-89bfaa42-2_arm_xscale.ipk
> Installing libubus-lua (2018-10-06-221ce7e7-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/libubus-lua_2018-10-06-221ce7e7-1_arm_xscale.ipk
> Installing luci-lib-jsonc (git-19.180.62908-a04e22e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-lib-jsonc_git-19.180.62908-a04e22e-1_arm_xscale.ipk
> Installing liblucihttp0 (2019-06-05-91c01c3c-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/liblucihttp0_2019-06-05-91c01c3c-1_arm_xscale.ipk
> Installing liblucihttp-lua (2019-06-05-91c01c3c-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/liblucihttp-lua_2019-06-05-91c01c3c-1_arm_xscale.ipk
> Installing luci-base (f138fc93-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-base_f138fc93-1_arm_xscale.ipk
> Installing libiwinfo-lua (2019-06-12-1372f47e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/targets/kirkwood/generic/packages/libiwinfo-lua_2019-06-12-1372f47e-1_arm_xscale.ipk
> Installing luci-mod-status (git-19.180.62908-a04e22e-1) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/luci/luci-mod-status_git-19.180.62908-a04e22e-1_arm_xscale.ipk
> Installing luci-mod-system (git-19.180.62908-a04e22e-1) to root...
[... 28 lines cut ...]
> Configuring luci-base.
> Configuring luci-mod-system.
> Configuring luci-theme-bootstrap.
> Configuring luci-mod-status.
> Configuring luci-app-firewall.
> Configuring luci-proto-ppp.
> Configuring luci-mod-network.
> Configuring luci-mod-admin-full.
> Configuring luci-proto-ipv6.
> Configuring uhttpd.
> Configuring rpcd-mod-rrdns.
> Configuring luci.
> 
> opkg install lua
> Installing lua (5.1.5-4) to root...
> Downloading http://downloads.openwrt.org/snapshots/packages/arm_xscale/base/lua_5.1.5-4_arm_xscale.ipk
> Configuring lua.
> root@OpenWrt:~#

Grettings! Thanks for the tip. It worked here, but in a different order:
1. opkg remove luci --force-removal-of-dependent-packages
2. opkg remove lua5.3 --force-removal-of-dependent-packages --force-depends
3. opkg update
3. opkg install lua
4. opkg install luci

hnyman, 2019-06-30:
I will close this, as @rmilecki already pushed the fix to the main Openwrt sources with
https://github.com/openwrt/openwrt/commit/17ae3eb9ff9dd96dc320dbe1b110e667b8b54c2a

Once the buildbot completes building the packages into the snapshot download repo, the issue should be fixed. (It will take some hours. so far about half of the targets have been built)

closed 2019-06-30 by hnyman

caimaojack, 2024-03-07:
我觉得没那么复杂，我也遇到过类似问题，应该是文本换行符引起的，把Windows(CR LF),切换成Unix(LF)就可以了。