openwrt/luci #2812: [feature suggestion] provide an option to save/restore WAN profiles for travel context

n8v8R, 2019-07-01:
* trunk
____

utilizing the router for travel and thus frequently connecting with different means of upstream WAN there is currently no option in LuCI to store WAN connectivity profiles. For the time being this can be worked around by manually (though ssh) saving/restoring WAN settings.

Imho it would be handy if LuCI would cater for saving WAN profiles and restoring such when needed in the travel context. Supposedly it would require to extract the WAN settings from /etc/config/network on the saving part and rewriting that portion/section when restoring.

(crude) layout mock-up (material theme)
[image]

jow-, 2019-07-01:
How is this supposed to work with other interfaces? I don‘t think it makes sense to tie that to wan. If at all, the profile thing schould be global abd probably not part of the interface settings

n8v8R, 2019-07-01:
Not sure why it needs to be tied in with other ifaces, e.g. lan, guest etc?

When restoring it supposedly would rewrite the WAN section in /etc/network/config and restart the WAN iface. That is what I currently do via ssh and just thought that it might be handy as a LuCI feature.

Please feel at liberty to close this if you are not in favour.

feckert, 2019-07-01:
You could add this to a user defined Command
https://github.com/openwrt/luci/tree/master/applications/luci-app-commands

n8v8R, 2019-07-01:
Would have to write a parser for extracting and saving the WAN section, loading and rewriting that section respectively.

n8v8R, 2019-07-01:
closing, since the number of devices being used in travel context is likely negligible

closed 2019-07-01 by n8v8R