openwrt/luci #2816: One byte missed when uploading upgrade image file with chrome

srdgame, 2019-07-03:
I am working on OpenWRT-19.07 branch.  I have my device working with 19.07. 

We got device failed to boot with image built today, caused by incorrect image file been flashed to device.

We found this only happened with **Chrome** and the "keep settings" option **off**.  And the image file save to /tmp in device only missed one byte:
```lua
	local fp
	local fp_len = 0
	http.setfilehandler(
		function(meta, chunk, eof)
			if not fp and meta and meta.name == "image" then
				fp = io.open(image_tmp, "w")
				fp_len = 0
			end
			if fp and chunk then
				fp_len = fp_len + string.len(chunk)
				fp:write(chunk)
			end
			if fp and eof then
				os.execute("echo file length: "..fp_len.." > /dev/console")
				fp:close()
			end
		end
	)

```

```
file length: 18021887
file length: 18021888
```

*Tested with Chrome in both Windows 7 & Debian testing, both have this issue
* FireFox & IE have no such issue.
* "Keep settings" must be unchecked to reproduce this bug
* Wireshark says the file length posted by Chrome is 18021888 when bug happens

srdgame, 2019-07-03:
* Changing the filename to not be "openwrt-sunxi-cortexa7-sun8i-h3-tlink-x1-squashfs-sdcard.img.gz" will not have this issue
* Using the uncompressed image openwrt-sunxi-cortexa7-sun8i-h3-tlink-x1-squashfs-sdcard.img has no such issue

[openwrt-sunxi-cortexa7-sun8i-h3-tlink-x1-squashfs-sdcard.img.gz](https://github.com/openwrt/luci/files/3353696/openwrt-sunxi-cortexa7-sun8i-h3-tlink-x1-squashfs-sdcard.img.gz)

jow-, 2019-07-03:
Can you provide a HAR archive of the failing image upload?

srdgame, 2019-07-03:
The wireshark capture file is too big to upload, I can email u it.

srdgame, 2019-07-03:
@jow what is a HAR archive?  I am sorry that I do not know about this.

jow-, 2019-07-03:
You can export that from the chrome developer tools when right clicking on the HTTP request in the network tab

srdgame, 2019-07-03:
[172.30.11.133.har.zip](https://github.com/openwrt/luci/files/3353742/172.30.11.133.har.zip)

srdgame, 2019-07-03:
@jow the wireshark capture has been emailed to you, via "dirk.chang-at-thingsroot.com"

srdgame, 2019-07-03:
Chrome Version:
Version 73.0.3683.75 (Developer Build) built on Debian buster/sid, running on Debian 10.0 (64-bit)
Version 0.69.27 Chromium: 75.0.3770.100 64bits (Brave mod? Win7) 
Version 75.0.3770.100 64bits (Google Chrome Win7)

srdgame, 2019-07-03:
The one byte missed via Win7 Chrome ( or Brave) is always the same byte.

The one byte missed via Linux Chrome  is always the same byte,  but not the same byte with Win7.

The win10 have the same issue and missed the same byte as Win7

@jow-

hnyman, 2019-07-03:
I am suddenly stumbling into the same behaviour when trying to upload firmware image with Firefox 68.0 to Openwrt 18.06.
LuCI openwrt-18.06 branch (git-19.181.61051-9bab4ba) / OpenWrt 18.06-SNAPSHOT r7810-b84f761d91 

scp:ing the image works naturally ok, but via LuCI the transferred image is one byte short.

```
root@router2:~# ls -lt /tmp/
-rw-r--r--    1 root     root       7668507 Jul  3 20:26 firmware.img
-rw-r--r--    1 root     root             0 Jul  3 20:26 sysupgrade.meta
-rw-r-----    1 root     root           184 Jul  3 20:26 0.db
-rw-r--r--    1 root     root       7668508 Jul  3 20:25 WNDR3700v2-master-r10387-c3e31b6a9b-20190703-2001-sqfs-sysupgrade.bin
```

I have had not trouble this far, and the upload via LuCI with the same browser went ok one hour ago when sysupgrading ipq806x/R7800 from 18.06 to master.  (Actually, the same versions: from owrt1806-r7810-b84f761d91 to master-r10387-c3e31b6a9b)

My previous attempt had a different sha256 and after reboot the sha256 has changed, so apparently a different byte is missing from the transfer.  On consecutive transfers, the sha256 hash remains same, so the error is repeatable (but did change after rebbot).

hnyman, 2019-07-03:
Not much help for debugging, but for me the difference seems to be in the middle of the file:

```
OK via SCP:
002cae40  c9 dc 4c 38 0d 0a 2c b7 d3 fb 67 19 50 5d 7d 6f c3 39 90 43 a5 09 c1 95

wrong via LuCI & uhttpd
002cae40  c9 dc 4c 38 0d 0a b7 d3 fb 67 19 50 5d 7d 6f c3 39 90 43 a5 09 c1 95 a3
```

This is from Windows 10 1903 using "keep settings" (unlike OP) and using https in the router.

(I tested with Microsoft Edge and that transferred the file ok.)

Disabling HTTPS openssl (via renaming /lib/ libustream-ssl.so) and trying HTTP instead of HTTPS does not help.

hnyman, 2019-07-03:
wireshark shows me that on the PC side that data is missing from middle of a TCP packet. PC thinks that the packet transferred ok. "7a" is missing from the uhttpd transferred file.

```
Wireshark in PC about the packet payload:

1b70                                 4b 88 ff b4 20 91
1b80   df 84 d1 02 01 77 d3 86 6c 0d 0a 7a cb 15 7c e5
1b90   5e 15 3e 

router file / Correct file via SCP:
00214638  4b 88 ff b4 20 91 df 84 d1 02 01 77 d3 86 6c 0d 0a 7a cb 15 7c e5 5e 15

router file / Wrong file via uhttpd:
00214638  4b 88 ff b4 20 91 df 84 d1 02 01 77 d3 86 6c 0d 0a cb 15 7c e5 5e 15 3e
```
 
Based on the wireshark package data, all packets have had the standard amount of 26180 bytes of payload, and no special protocol exchange related to this packet:

```
361	7.315122	192.168.1.2	192.168.1.188	TCP	60	[TCP Window Update] 80 → 62290 [ACK] Seq=2 Ack=2174733 Win=3384 Len=0
362	7.315183	192.168.1.188	192.168.1.2	TCP	26334	62290 → 80 [PSH, ACK] Seq=2174733 Ack=2 Win=1026 Len=26280 [TCP segment of a reassembled PDU]
363	7.319066	192.168.1.2	192.168.1.188	TCP	60	80 → 62290 [ACK] Seq=2 Ack=2201013 Win=99 Len=0
364	7.333308	192.168.1.2	192.168.1.188	TCP	60	[TCP Window Update] 80 → 62290 [ACK] Seq=2 Ack=2201013 Win=3384 Len=0
365	7.333365	192.168.1.188	192.168.1.2	TCP	26334	62290 → 80 [ACK] Seq=2201013 Ack=2 Win=1026 Len=26280 [TCP segment of a reassembled PDU]
366	7.336878	192.168.1.2	192.168.1.188	TCP	60	80 → 62290 [ACK] Seq=2 Ack=2227293 Win=99 Len=0

```

jow-, 2019-07-03:
`7a` follows `0a` (a line feed) which may cause state transitions in MIME multipart parsing. Please try if reverting https://github.com/openwrt/luci/commit/15608fd2e5283fd6dd3ee4e9bf783029d0cae8c3 helps.

hnyman, 2019-07-03:
Might be, as also in my first example above the missing byte 2c was after 0a.
hmmm.... Actually, in both cases the missing byte was after "0d 0a" = CR LF, not just LF

I will test that. 
Simply re-compiling and replacing these?
liblucihttp - 2019-06-05-91c01c3c-1
liblucihttp-lua - 2019-06-05-91c01c3c-1

(I always look at the sha256 checksum in sysupgrade, and I have not seen this earlier, so in any case this is likely something rather new.)

hnyman, 2019-07-03:
hmmm. 
In your lucihttp change of 2019-06-05, you specifically evaluate for CR, so the reason might well be the combination of CR LF just at some boundary.

In any case, I uploaded the failing image to dropbox, in case you want to test the file transfer yourself.
WNDR3700v2-master-r10387-c3e31b6a9b-20190703-2001-sqfs-sysupgrade.bin
https://www.dropbox.com/sh/s44q2lsnl9nnas1/AADLM6XzGlOpJHee9pBkZmlha?dl=0

sha256 hash:  d745a8f2f846459e5f7fceee2d20f0502a2a54551c1da9eb96857d61943f8b84

hnyman, 2019-07-03:
After reverting 0dd8878 and 115c4e3  (the 18.06 versions of the lucihttp version bump commits), the file transfer seems to go ok.

jow-, 2019-07-04:
Please try bumping lucihttp to https://github.com/jow-/lucihttp/commit/730a46f26f7665ce7d74ab282d5aecadd735d456

hnyman, 2019-07-04:
Thanks. This seems to fix things.

I compiled and tested the new version both with 18.06 and master. 
I sysupgraded with this from 18.06 to master with the same image that was problematic yesterday, and then tested the new version also in master for further sysupgrades. Seems to be ok also there.

A bit sad that the old version is in 18.06.4 images. That may be problematic in sysupgrading to the next release to 19.07. Once the buildbot has compiled the fixed version of lucihttp, you might maybe add this as a "known issues" section in the 18.06.4 release notes.

closed 2019-07-05 by jow-

srdgame, 2019-07-10:
will test it :-)

commit 66c8ebfe (2019-07-05): lucihttp: update to latest Git HEAD

a34a17d src: allow overriding buffer size from cli in multipart tester
730a46f lib: fix potentially lost bytes in boundary parsing across buffer limits
8734af2 lib: add buffer tracing to multipart parser
913051b src: add file dump option to multipart test utility
c419539 src: allow specifying custom buffer sizes in multipart testcases
f6e0564 lib: fix handling of empty multipart fields

Fixes: #2816
commit f166eb6d (2019-07-05): lucihttp: update to latest Git HEAD

a34a17d src: allow overriding buffer size from cli in multipart tester
730a46f lib: fix potentially lost bytes in boundary parsing across buffer limits
8734af2 lib: add buffer tracing to multipart parser
913051b src: add file dump option to multipart test utility
c419539 src: allow specifying custom buffer sizes in multipart testcases
f6e0564 lib: fix handling of empty multipart fields

Fixes: #2816

(cherry picked from commit 66c8ebfeffcffdcfcdf64888b7685dca26f82ff5)
commit 0593f935 (2019-07-05): lucihttp: update to latest Git HEAD

a34a17d src: allow overriding buffer size from cli in multipart tester
730a46f lib: fix potentially lost bytes in boundary parsing across buffer limits
8734af2 lib: add buffer tracing to multipart parser
913051b src: add file dump option to multipart test utility
c419539 src: allow specifying custom buffer sizes in multipart testcases
f6e0564 lib: fix handling of empty multipart fields

Fixes: #2816

(cherry picked from commit 66c8ebfeffcffdcfcdf64888b7685dca26f82ff5)