openwrt/luci #2820: advanced_reboot trims the version of installed linux shown

rezad1393, 2019-07-04:
I see this in my luci interface:

 1
Alternative
OpenWrt (Linux 4.14.95)
2
Current
OpenWrt (Linux 4.14.13) 

but the current linux installed is Linux openwrt 4.14.131 
this is on the latest version of openwrt 18.06.04
and I noticed it today.
dont know if it was present before.

dibdot, 2019-07-05:
@stangri

stangri, 2019-07-05:
@rezad1393 which version of luci-app-advanced-reboot do you have?

rezad1393, 2019-07-06:
latest version.
why?
do you get a different result on your openwrt stable installation?
luci-app-advanced-reboot - git-19.186.23306-0593f93-33

hnyman, 2019-07-06:
> latest version.
> why?
> do you get a different result on your openwrt stable installation?

"Latest"? on master?  or on 18.06.x or 19.07 ?

On master the version gets show properly. Possibly the fix has not been backported to older branches. (master and 19.07 seem to have version 39 of the package, while 18.06 still has version 33.)

From master:
```
 OpenWrt SNAPSHOT, r10402-ee66fe4ea9
 -----------------------------------------------------
root@router3:~# opkg list-installed *adva*
luci-app-advanced-reboot - git-19.187.53290-3a24c2d-39
```
```
Action
1
Current
OpenWrt (Linux 4.19.56)
2
Alternative
OpenWrt (Linux 4.14.131) 
```

rezad1393, 2019-07-06:
18.06.x

stangri, 2019-07-06:
submitted PR for stable: https://github.com/openwrt/luci/pull/2831

closed 2019-07-07 by hnyman