openwrt/luci #2833: Trunk "Bad Gateway"

jammmet, 2019-07-07:
Compiled trunk, on ramips MT7620
Running uhttpd on port 8080 - always worked before (up till last month some time).
Now I get "Bad Gateway"
`dmesg` shows:

```
[  143.408870] do_page_fault(): sending SIGSEGV to luci for invalid read access from 00000008
[  143.425151] epc = 77e6e003 in liblua.so.5.1.5[77e60000+2e000]
[  143.431082] ra  = 77e6e09f in liblua.so.5.1.5[77e60000+2e000]
```

jow-, 2019-07-07:
Seems Lua is segfaulting. Not sure if this is related to LuCI at all. You need to obtain a backtrace using gdb.

On the router:

```
ulimit -c unlimited
/www/cgi-bin/luci
```

Scp the resulting coredump to your development system, run `./scripts/remote-gdb /path/to/corefile ./build_dir/target-*/lua-5.1.5/src/lua5.1`.

In the gdb shell type `bt full` and provide the output here.

jammmet, 2019-07-07:
Thanks @jow- output:
```
(gdb) bt full
#0  0x77ec1003 in luaH_getstr () from /home/builder/oneboxx-newifi/scripts/../staging_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/liblua.so.5.1.5
No symbol table info available.
#1  0x77ec109f in luaH_get () from /home/builder/oneboxx-newifi/scripts/../staging_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/liblua.so.5.1.5
No symbol table info available.
Backtrace stopped: frame did not save the PC
```

jow-, 2019-07-07:
That looks like a crash somewhere deep down in Lua itself. Can you reproduce this on a normal OPenWrt snapshot build?

jammmet, 2019-07-12:
I can't unfortunately - regular build works fine
Are there any missing packages?
```
base-files - 203-r10494-e05310b
block-mount - 2019-03-28-ff1ded63-5
busybox - 1.31.0-1
ca-bundle - 20190110-1
chat - 2.4.7.git-2019-05-25-2
comgt - 0.32-32
curl - 7.65.1-1
dnsmasq - 2.80-13
dosfstools - 4.1-3
e2fsprogs - 1.44.5-1
firewall - 2019-01-02-70f8785b-2
fstools - 2019-03-28-ff1ded63-5
fwtool - 1
getrandom - 2019-06-16-4df34a4d-2
hostapd-common - 2018-12-02-c2c6c01b-6
ip-tiny - 5.1.0-1
ip6tables - 1.8.3-1
iptables - 1.8.3-1
iptables-mod-conntrack-extra - 1.8.3-1
iptables-mod-ipopt - 1.8.3-1
iptables-mod-ipsec - 1.8.3-1
iw - 5.0.1-1
iwinfo - 2019-06-12-1372f47e-1
jshn - 2019-06-16-ecf56174-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.14.132-1-149a94cb04aa536418c6152a034ddf8c
kmod-cfg80211 - 4.14.132+4.19.32-1-2
kmod-crypto-acompress - 4.14.132-1
kmod-crypto-aead - 4.14.132-1
kmod-crypto-authenc - 4.14.132-1
kmod-crypto-cbc - 4.14.132-1
kmod-crypto-cmac - 4.14.132-1
kmod-crypto-crc32 - 4.14.132-1
kmod-crypto-crc32c - 4.14.132-1
kmod-crypto-ctr - 4.14.132-1
kmod-crypto-deflate - 4.14.132-1
kmod-crypto-des - 4.14.132-1
kmod-crypto-echainiv - 4.14.132-1
kmod-crypto-gcm - 4.14.132-1
kmod-crypto-gf128 - 4.14.132-1
kmod-crypto-ghash - 4.14.132-1
kmod-crypto-hash - 4.14.132-1
kmod-crypto-hmac - 4.14.132-1
kmod-crypto-iv - 4.14.132-1
kmod-crypto-manager - 4.14.132-1
kmod-crypto-md5 - 4.14.132-1
kmod-crypto-null - 4.14.132-1
kmod-crypto-pcompress - 4.14.132-1
kmod-crypto-rng - 4.14.132-1
kmod-crypto-seqiv - 4.14.132-1
kmod-crypto-sha1 - 4.14.132-1
kmod-crypto-sha256 - 4.14.132-1
kmod-crypto-wq - 4.14.132-1
kmod-eeprom-93cx6 - 4.14.132-1
kmod-fs-exfat - 4.14.132+2018-11-06-2ac42ac4-3
kmod-fs-ext4 - 4.14.132-1
kmod-fs-f2fs - 4.14.132-1
kmod-fs-vfat - 4.14.132-1
kmod-gpio-button-hotplug - 4.14.132-2
kmod-ifb - 4.14.132-1
kmod-ip6tables - 4.14.132-1
kmod-ipsec - 4.14.132-1
kmod-ipsec4 - 4.14.132-1
kmod-ipsec6 - 4.14.132-1
kmod-ipt-conntrack - 4.14.132-1
kmod-ipt-conntrack-extra - 4.14.132-1
kmod-ipt-core - 4.14.132-1
[... 227 lines cut ...]
uclient-fetch - 2019-05-30-3b3e368d-1
uhttpd - 2019-06-16-91fcac34-3
uhttpd-mod-ubus - 2019-06-16-91fcac34-3
urandom-seed - 1.0-1
urngd - 2019-06-17-c057e177-1
usbreset - 5
usign - 2015-07-04-ef641914-1
wget - 1.20.3-1
wireless-regdb - 2017-10-20-4343d359
wireless-tools - 29-6
wpad-basic - 2018-12-02-c2c6c01b-6
zlib - 1.2.11-3
zoneinfo-africa - 2019b-1
zoneinfo-asia - 2019b-1
zoneinfo-atlantic - 2019b-1
zoneinfo-australia-nz - 2019b-1
zoneinfo-core - 2019b-1
zoneinfo-europe - 2019b-1
zoneinfo-india - 2019b-1
zoneinfo-northamerica - 2019b-1
zoneinfo-pacific - 2019b-1
zoneinfo-poles - 2019b-1
zoneinfo-simple - 2019b-1
zoneinfo-southamerica - 2019b-1
```

jow-, 2019-07-12:
I don't think it is related to missing packages. Its probably something involving compiler optimization, nonstandard CFLAGS or similar.

Whats the output of `./scripts/diffconfig.sh` ?

jammmet, 2019-07-12:
```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7620=y
CONFIG_TARGET_ramips_mt7620_DEVICE_zbtlink_zbt-we826-16m=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BINUTILS_USE_VERSION_2_31_1 is not set
CONFIG_BINUTILS_USE_VERSION_2_32=y
CONFIG_BINUTILS_VERSION="2.32"
CONFIG_BINUTILS_VERSION_2_32=y
CONFIG_BUILD_PATENTED=y
# CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_NEWER is not set
CONFIG_BUSYBOX_CONFIG_FEATURE_MOUNT_LABEL=y
CONFIG_BUSYBOX_CONFIG_HDPARM=y
CONFIG_BUSYBOX_CONFIG_LSUSB=y
CONFIG_BUSYBOX_CONFIG_NOHUP=y
# CONFIG_BUSYBOX_CONFIG_SWAPOFF is not set
# CONFIG_BUSYBOX_CONFIG_SWAPON is not set
CONFIG_BUSYBOX_CONFIG_VOLUMEID=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
CONFIG_GCC_USE_EMBEDDED_PATH_REMAP=y
# CONFIG_GCC_USE_VERSION_7 is not set
CONFIG_GCC_USE_VERSION_9=y
CONFIG_GCC_VERSION="9.1.0"
CONFIG_GCC_VERSION_9=y
CONFIG_INCLUDE_CONFIG=y
# CONFIG_KERNEL_AIO is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_FANOTIFY is not set
# CONFIG_KERNEL_FHANDLE is not set
# CONFIG_KERNEL_LXC_MISC is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_SECCOMP is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_NGINX_HEADERS_MORE=y
CONFIG_NGINX_HTTP_ACCESS=y
CONFIG_NGINX_HTTP_AUTH_BASIC=y
CONFIG_NGINX_HTTP_AUTOINDEX=y
CONFIG_NGINX_HTTP_BROWSER=y
CONFIG_NGINX_HTTP_CACHE=y
CONFIG_NGINX_HTTP_CHARSET=y
CONFIG_NGINX_HTTP_EMPTY_GIF=y
CONFIG_NGINX_HTTP_FASTCGI=y
CONFIG_NGINX_HTTP_GEO=y
CONFIG_NGINX_HTTP_GZIP=y
CONFIG_NGINX_HTTP_LIMIT_CONN=y
CONFIG_NGINX_HTTP_LIMIT_REQ=y
CONFIG_NGINX_HTTP_MAP=y
CONFIG_NGINX_HTTP_MEMCACHED=y
CONFIG_NGINX_HTTP_PROXY=y
CONFIG_NGINX_HTTP_REFERER=y
CONFIG_NGINX_HTTP_REWRITE=y
CONFIG_NGINX_HTTP_SCGI=y
CONFIG_NGINX_HTTP_SPLIT_CLIENTS=y
CONFIG_NGINX_HTTP_SSI=y
CONFIG_NGINX_HTTP_UPSTREAM_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_IP_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_KEEPALIVE=y
CONFIG_NGINX_HTTP_UPSTREAM_LEAST_CONN=y
CONFIG_NGINX_HTTP_USERID=y
[... 235 lines cut ...]
CONFIG_PACKAGE_wget=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_wpad-mini=m
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zoneinfo-africa=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-atlantic=y
CONFIG_PACKAGE_zoneinfo-australia-nz=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PACKAGE_zoneinfo-europe=y
CONFIG_PACKAGE_zoneinfo-india=y
CONFIG_PACKAGE_zoneinfo-northamerica=y
CONFIG_PACKAGE_zoneinfo-pacific=y
CONFIG_PACKAGE_zoneinfo-poles=y
CONFIG_PACKAGE_zoneinfo-simple=y
CONFIG_PACKAGE_zoneinfo-southamerica=y
CONFIG_PHP7_FILTER=y
CONFIG_PHP7_LIBXML=y
CONFIG_PHP7_SYSTEMTZDATA=y
# CONFIG_POSTFIX_PCRE is not set
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_OPENSSL_ENGINE is not set
```

8a9sd, 2019-07-13:
For me on trunk building for ath79 with the same type of error, switching back to gcc8 or gcc7 solved the lua issue. So seems like it's related to building on gcc9 right now. Hope this helps.

jow-, 2019-07-14:
This is not a LuCI issue then. If you use nonstandard compiler versions you're on your own.

closed 2019-07-14 by jow-

jammmet, 2019-07-14:
Can confirm. A clean build on GCC 7 works fine.

avbohemen, 2020-03-19:
Issue does not occur for me anymore with luci - git-20.033.32700 / OpenWrt SNAPSHOT, r12162 / gcc 9.3