openwrt/luci #2834: LuCI version string broken down in March-April

hnyman, 2019-07-07:
cc @jow- 

Looks like we have a LuCI versioning problem for luci-base.

I noticed that my master and 19.07 builds show always the exact LuCI version as "LuCI Master (f138fc93)" instead of the expected "openwrt-18.06 branch (git-19.170.32094-4d6d8bc)" style of version.

Openwrt 18.06 builds show the version ok with the date-based string, while master and 19.07 show always that "LuCI Master (f138fc93)".  And always that same hash f138fc93

The wrong string is contained in the  /usr/lib/lua/luci/version.lua included in the firmware image
```
luciname    = "LuCI Master"
luciversion = "f138fc93"
```

I checked from buildbot and also in https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/luci/luci-base_f138fc93-1_arm_cortex-a15_neon-vfpv4.ipk the string is wrong. So, the fault affects also the buildbot.

That is actually visible in the download dir already in the name of the luci-base package:

luci-app-wireguard_git-19.187.53290-3a24c2d-1_all.ipk | 3.7 KB | Sun Jul  7 09:41:10 2019
luci-app-wol_git-19.187.53290-3a24c2d-1_all.ipk | 2.2 KB | Sun Jul  7 08:35:07 2019
luci-base_f138fc93-1_arm_cortex-a15_neon-vfpv4.ipk | 128.1 KB | Sun Jul  7 09:02:43 2019
luci-i18n-adblock-es_git-19.187.53290-3a24c2d-1_all.ipk | 5.4 KB | Sun Jul  7 09:02:45 2019
luci-i18n-adblock-it_git-19.187.53290-3a24c2d-1_all.ipk | 2.7 KB | Sun Jul  7 09:02:45 2019
 
I looked at my old monthly archives, and the end of February build from 25.2.2019 has the correct string "git-19.056.59493-9c3132b", but end of April build already has the wrong string.

I can't figure out a possible culprit change in end-of-Feb to end-of-April by looking at the source code. Nothing obvious jumps out.

hnyman, 2019-07-07:
I figured it out by brute force testing, but it might have been seen also from the sources:

The culprit is:
"luci-base: luasrcdiet: switch to scm clone and fix tarball name"
It changed the luci-base Makefile to contain git download for luasrcdiet. That apparently overrides the main git repo version, and the version "f138fc93" shown is actually the git hash of luasrcdiet
https://github.com/openwrt/luci/commit/faa5f0eacd2b2c90c33a75ee1e7fd3ec776601af
```
PKG_SOURCE_URL:=https://github.com/jirutka/luasrcdiet.git
PKG_SOURCE_VERSION:=f138fc9359821d9201cd6b57cfa2fcbed5b9af97
```
@jow- 
That commit could be reverted, I guess, but do you want to figure out a better way to preserve the git version from "upstream" main luci.mk ?

Testing:
```
perus@ub1904:/Openwrt/wrt3200/feeds/luci$ git reset --hard 5624b36fa7a5f
HEAD is now at 5624b36fa luci-mod-network: fix page refresh after interface reconnect
perus@ub1904:/Openwrt/wrt3200$ tail -n 2 build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/luci-base/src/version.lua 
luciname    = "LuCI Master"
luciversion = "git-19.098.22890-5624b36"
```

```
perus@ub1904:/Openwrt/wrt3200/feeds/luci$ git reset --hard faa5f0eacd2b
HEAD is now at faa5f0eac luci-base: luasrcdiet: switch to scm clone and fix tarball name
perus@ub1904:/Openwrt/wrt3200$ tail -n 2 build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/luci-base/src/version.lua 
luciname    = "LuCI Master"
luciversion = "f138fc93"
```

hnyman, 2019-07-07:
As far as I can figure out, the mechanics go this way:

* we now define PKG_SOURCE_VERSION for luasrcdiet in luci-base Makefile
* That hash gets shortened for PKG_VERSION in include/download.mk
 https://github.com/openwrt/openwrt/blob/b812a7fa689a5a6dea3a9047108e2f1d1ec1cf36/include/download.mk#L14
* As we now have a PKG_VERSION, the git-based version detection logic in luci.mk never gets activated, as it is behind conditional evaluation "PKG_VERSION?=" and we now have PKG_VERSION (from luasrcdiet source download logic)

That PKG_SOURCE_VERSION might be avoided by using the codeload approach originally proposed by @diizzyy in #2522 

Other approach might be to properly separate luasrcdiet from luci-base

jow-, 2019-07-07:
Thanks I reverted the offending commit. Adding a hostonly build for srcdiet is probably the best solution, but I don't have time for that atm.

commit 8055ed4c (2019-07-07): Revert "luci-base: luasrcdiet: switch to scm clone and fix tarball name"

This reverts commit faa5f0eacd2b2c90c33a75ee1e7fd3ec776601af.

Fixes: #2834
commit 5566b636 (2019-07-07): Revert "luci-base: luasrcdiet: switch to scm clone and fix tarball name"

This reverts commit faa5f0eacd2b2c90c33a75ee1e7fd3ec776601af.

Fixes: #2834

(cherry picked from commit 8055ed4c469516932a7abceed41b13cc7a11de35)
closed 2019-07-07 by jow-