openwrt/luci #2846: Trunk LuCI firewall page does not load

gwlim, 2019-07-09:
To reproduce
-Install openwrt trunk with luci
-Go to firewall page

```
this.ifname.charAt is not a function
  at getType (http://192.168.1.7/luci-static/resources/network.js:248:585)
  at deviceSort (http://192.168.1.7/luci-static/resources/network.js:97:114)
  at getDevices (http://192.168.1.7/luci-static/resources/network.js:239:4)
  at renderZone (http://192.168.1.7/luci-static/resources/tools/widgets.js:12:172)
  at renderWidget (http://192.168.1.7/luci-static/resources/tools/widgets.js:20:178)
```
I checked zone.js, is it wise to use offload to decide whether to render the page?

jow-, 2019-07-09:
Canont reproduce it here on a clean build. Please provide the following command outputs:

```
ubus call network.wireless status
ubus call luci netdevs
ubus call luci ifaddrs
ubus call luci boardjson
ubus call network.interface dump
ubus call network.device status
```

gwlim, 2019-07-09:
It is a standard firefox.
The only other non-standard thing I did was I selected 
LUCI_SRCDIET=y
I cloned from trunk, ./scripts/feeds update -a && ./scripts/feeds install -a
Then select TP-Link WR1043NDv1 and LuCI metapackage.
That is all

jow-, 2019-07-09:
I need the output of the ubus commands above to figure out the root cause. You can also send them via mail to me if you prefer that.

gwlim, 2019-07-09:
```
{
	"radio0": {
		"up": false,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": true,
		"config": {
			"channel": "11",
			"hwmode": "11g",
			"path": "platform/ath9k",
			"htmode": "HT20"
		},
		"interfaces": [
			
		]
	},
	"radio1": {
		"up": true,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": false,
		"config": {
			"channel": "11",
			"hwmode": "11g",
			"path": "platform/ahb/180c0000.wmac",
			"htmode": "HT20"
		},
		"interfaces": [
			
		]
	}
}
{
	"eth0.2": {
		"mac": "Removed",
		"type": 1,
		"name": "eth0.2",
		"up": true,
		"mtu": 1500,
		"qlen": 1000
	},
	"lo": {
		"mac": "00:00:00:00:00:00",
		"type": 772,
		"name": "lo",
		"up": true,
		"mtu": 65536,
		"qlen": 1000
	},
	"br-lan": {
		"mac": "Removed",
		"type": 1,
		"stp": false,
		"up": true,
		"ports": [
			"eth0.1"
		],
		"bridge": true,
		"name": "br-lan",
		"id": "Removed",
		"mtu": 1500,
		"qlen": 1000
	},
	"eth0": {
		"mac": "Removed",
		"type": 1,
		"name": "eth0",
[... 916 lines cut ...]
			"tx_compressed": 0,
			"multicast": 0,
			"rx_length_errors": 0,
			"tx_dropped": 0,
			"rx_bytes": 12128,
			"rx_missed_errors": 0,
			"tx_errors": 0,
			"rx_compressed": 0,
			"rx_over_errors": 0,
			"tx_fifo_errors": 0,
			"rx_crc_errors": 0,
			"rx_packets": 173,
			"tx_heartbeat_errors": 0,
			"rx_dropped": 0,
			"tx_aborted_errors": 0,
			"tx_packets": 173,
			"rx_errors": 0,
			"tx_bytes": 12128,
			"tx_window_errors": 0,
			"rx_fifo_errors": 0,
			"tx_carrier_errors": 0
		}
	}
}
```

jow-, 2019-07-09:
Thank you for the data - that helped a lot to find the root cause.

closed 2019-07-09 by jow-

gwlim, 2019-07-09:
Hi @jow-  can I also trouble you to look at this?
https://bugs.openwrt.org/index.php?do=details&task_id=2364&order=dateopened&sort=desc
It is a very serious bug and I tested it using a cleanly build trunk with LuCI

commit ffdafd48 (2019-07-09): luci-base: properly handle getWifiNetidBySid() return value

The getWifiNetidBySid() helper returns an array containing the network id
and the corresponding radio name, not just the id.

Fixes: #2846.