openwrt/luci #2847: [potential bug] luci-proto-ppp requesting package that is not available

n8v8R, 2019-07-09:
* 19.07
* luci-proto-ppp  git-19.190.43457-7728bf3-1.0
* kmod-pppoa Version: 4.14.131-1.0 (is installed)
____
that package is not available from the feed stream (and does not seem required by luci-proto-ppp)

[image]
___
[image]

jow-, 2019-07-10:
https://downloads.openwrt.org/releases/packages-18.06/x86_64/base/ppp-mod-pppoa_2.4.7-12_x86_64.ipk

closed 2019-07-10 by jow-

n8v8R, 2019-07-10:
Testing 19.07 and that is where it happens. Thought to let you know at least.

jow-, 2019-07-10:
There are no 19.07 binaries yet at all on the download server - this is maybe the reason.