openwrt/luci #2849: [luci-lib-jsonc] needs libjson-c.so.2 but should need libjson-c.so.4

riddicc, 2019-07-10:
Hi!

Somehow the package
luci-lib-jsonc (git-19.186.54187-cbc000b-1 )
needs libjson-c.so.2.
But: libjson-c provides libjson-c.so.4 (and not libjson-c.so.2).

Workaround:
I sym-linked /rom/usr/lib/libjson-c.so.2.0.2 to /usr/lib/libjson-c.so.2
giggle :slight_smile:

How can I fix it properly?

Thx.

Bye.

jow-, 2019-07-10:
Reinstall/recompile luci-lib-jsonc

closed 2019-07-10 by jow-

riddicc, 2019-07-10:
but it is the same with
luci-lib-jsonc_git-19.189.59078-9c5162d-1_mips_24kc.ipk

why do u provide a package, if it is unusable with a workaround?

or who is it who provides it?

jow-, 2019-07-10:
```
jow@j460 $ wget -qO - https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/luci/luci-lib-jsonc_git-19.189.59078-9c5162d-1_mips_24kc.ipk | \
    tar -Oxzf - ./data.tar.gz | tar -Oxzf - ./usr/lib/lua/luci/jsonc.so | strings | grep libjson-c
libjson-c.so.2
jow@j460 $ 
```

jow-, 2019-07-10:
Can you try an `opkg update; opkg install --force-reinstall libjson-c`? 
Note that this is potentially dangerous (opkg sometimes fails while replacing packages) so be sure how to recover (failsafe etc.) before trying this.

riddicc, 2019-07-10:
oops...
fixed...
but with the same "trick" i got the [libjson-c.so.4 some weeks ago](https://forum.openwrt.org/t/updates-at-2019-06-23-libjson-c/39477)...
is there a problem with the version numbers?

jow-, 2019-07-10:
There was a brief period where packages got mixed up in the repo a few weeks ago, maybe this caused your issue as well.

riddicc, 2019-07-10:
oh... i c...
so it is ok, to use opkg to do upgrades?
some say i should always make a new image and flash the device with it...
that would be too much stress for the flash memory and too much downtime for my taste... :)