openwrt/luci #2852: Firewall configuration cannot be saved from Luci

rankaiyx, 2019-07-11:
Firewall configuration cannot be saved from Luci 
openwrt version: 19.07 20190710
Hardware:CPU MT7620N   ROM 8MBytes   RAM 64MBytes
Can be changed from config file:  /etc/config/firewall
config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        option network 'lan'

jow-, 2019-07-11:
Please elaborate - what means "cannot be saved" ? Do you get an error?

rankaiyx, 2019-07-11:
> Please elaborate - what means "cannot be saved" ? Do you get an error?
1 select "ACCEPT" or "reject"
2 click "save and apply" 
3 "ACCEPT" or "reject" will become "please choose"
 After modifying the configuration file,  can see the changes from LUCI.
There's no information because that's the GUI, but it feels like it stops running for about 10 seconds.

rankaiyx, 2019-07-11:
[image]

rankaiyx, 2019-07-11:
There's no information about that in system log

jow-, 2019-07-11:
Please provide the exact version number reported at the index page.

rankaiyx, 2019-07-11:
> Please provide the exact version number reported at the index page.

MediaTek MT7620N ver:2 eco:3
OpenWrt SNAPSHOT r10459-1174b94bc9 / LuCI Master (git-19.190.57021-ffdafd4)
4.14.132

closed 2019-07-11 by jow-

fishxz, 2019-07-12:
still unable to apply firewall changes

OpenWrt SNAPSHOT r10495-ba5878f056 / LuCI Master (git-19.193.31361-7c9d845)

rankaiyx, 2019-07-13:
I modified the UI. JS file to follow JOW's new code, but the bug still exists.
Although there are two UI.JS files, one of which is under the /ROM directory and cannot be modified, its priority is lower than that of the file that can be modified. It should be possible not to modify this file.

commit 43043a46 (2019-07-11): luci-base: ui.js: fix UISelect.getValue()

Fixes: #2852
Fixes: ce04d7f04 ("luci-base: fix label-to-input association in js cbi maps")
commit 3dcb061d (2019-07-12): luci-base: ui.js: fix UISelect.setValue()

Fixes: #2852
Fixes: ce04d7f ("luci-base: fix label-to-input association in js cbi maps")