openwrt/luci #2856: Showing HTML code of single quote character instead of symbol in Disconnect button

YuriPet, 2019-07-12:
In Bootstrap and OpenWrt themes showing HTML code of single quote character instead of symbol in "Disconnect" button.
[image]
[image]

YuriPet, 2019-07-16:
Obviously, nobody is interested in correcting this error.
So I went bypass – instead "Single quote" or "Apostrophe" (U+0027), I used the "Modifier letter apostrophe" (U+02BC).

[image]

Closed!

closed 2019-07-16 by YuriPet

jow-, 2019-07-19:
> Obviously, nobody is interested in correcting this error.

I am, but didn't find time yet to look into it. As in interim solution, the use of another unicode character looks good to me :+1:

YuriPet, 2019-07-19:
But still this is a "temporary" solution.
I believe that it is still necessary to make corrections in the corresponding code.

jow-, 2019-07-19:
Of course, I just need to carefully review the code to not accidentally open any XSS holes.

pjuhen, 2019-08-26:
Same issue for me